appliedAI-Initiative/orb_slam_2_ros #2: Issue running orb_slam2_ros

aievangelist, 2019-01-30:
When I run orb_slam2_ros, I am able to get the image from the subscribed topic. e.g. /usb_cam/image_raw  but I get an error on the console saying: Map Point Vector is empty!. What's the cause of this error and what is the possible reason for it?

lennarthaller, 2019-01-30:
This message means that the monocular SLAM is not yet initialized. In order to get the depth information for each key point the the monocular SLAM uses parallax (translational motion) of the camera. The debug image in this case should display green lines instead of key points, indicating that the depth is not yet initialized.

Move the camera around until the system is initialized. This is noted with the message "New map created" and the SLAM getting position lock.

tnaderi, 2019-02-01:
Hi, my name is Taher Naderi. Thank you for sharing your codes. I have built the package. I do not know how I should run it. I have an Intel real sense D435 camera and I would like to run the package based on that. I would be grateful if you could let me know where I should start.
Thank you.

lennarthaller, 2019-02-03:
Hey,

In order to run the node with the D435 you'd need the intrinsic calibration parameters for this camera. You can obtain them with openCV and create new config files in orb_slam2/config for the newer RealSense.
In the launch file you probably need to adjust the topics to subscribe to for the D435.

Feel free to create a Pull request for the config files with parameters for the new camera!

tnaderi, 2019-02-04:
In my case it says that slam does not receive images. I tried openCV python file it gives me the parameters. But still I think something is wrong with subscribing images. Please help me in this regard.
Thank you

tnaderi, 2019-02-04:
https://docs.opencv.org/3.1.0/dc/dbb/tutorial_py_calibration.html

lennarthaller, 2019-02-04:
In the launch file you are using located in ros/launch you need to adjust the config file and the topic names to match your setup.

tnaderi, 2019-02-05:
I changed the launch file and it is working now. Thank you for your help.

lennarthaller, 2019-02-06:
Perfect. Closing this now.

closed 2019-02-06 by lennarthaller

juniorschen, 2020-08-31:
@lennarthaller 
Could you tell me if I can do mono slam with just a raspicamera?

ozumipl, 2022-08-11:
> I changed the launch file and it is working now. Thank you for your help.

Hi,
Currently I am having same problem and I did not understand how to solve problem. Here what I did:
I did run that: `roslaunch realsense2_camera rs_rgbd.launch `
After that I did run it: `roslaunch orb_slam2_ros orb_slam2_d435_mono.launch`

As an output log, I did obtain the following lines:
`... logging to /home/jetsonnano2/.ros/log/279dc6da-196e-11ed-a084-3413e863a5ce/roslaunch-jetsonnano2-desktop-24375.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://10.200.76.132:33919/

SUMMARY
========

PARAMETERS
 * /orb_slam2_mono/ORBextractor/iniThFAST: 20
 * /orb_slam2_mono/ORBextractor/minThFAST: 7
 * /orb_slam2_mono/ORBextractor/nFeatures: 2000
 * /orb_slam2_mono/ORBextractor/nLevels: 8
 * /orb_slam2_mono/ORBextractor/scaleFactor: 1.2
 * /orb_slam2_mono/camera_cx: 328.001068115
 * /orb_slam2_mono/camera_cy: 241.310317993
 * /orb_slam2_mono/camera_fps: 30
 * /orb_slam2_mono/camera_frame_id: camera_link
 * /orb_slam2_mono/camera_fx: 615.670715332
 * /orb_slam2_mono/camera_fy: 615.962158203
 * /orb_slam2_mono/camera_k1: 0.0
 * /orb_slam2_mono/camera_k2: 0.0
 * /orb_slam2_mono/camera_k3: 0.0
 * /orb_slam2_mono/camera_p1: 0.0
 * /orb_slam2_mono/camera_p2: 0.0
 * /orb_slam2_mono/camera_rgb_encoding: True
 * /orb_slam2_mono/load_calibration_from_cam: False
 * /orb_slam2_mono/load_map: False
 * /orb_slam2_mono/localize_only: False
 * /orb_slam2_mono/map_file: map.bin
 * /orb_slam2_mono/min_num_kf_in_map: 5
 * /orb_slam2_mono/pointcloud_frame_id: map
 * /orb_slam2_mono/publish_pointcloud: True
 * /orb_slam2_mono/publish_pose: True
 * /orb_slam2_mono/reset_map: False
 * /orb_slam2_mono/voc_file: /home/jetsonnano2...
 * /rosdistro: melodic
 * /rosversion: 1.14.13

NODES
  /
    orb_slam2_mono (orb_slam2_ros/orb_slam2_ros_mono)

ROS_MASTER_URI=http://10.200.76.132:11311/

process[orb_slam2_mono-1]: started with pid [24392]

ORB-SLAM2 Copyright (C) 2014-2016 Raul Mur-Artal, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

OpenCV version : 4.1.1
Major version : 4
Minor version : 1
Subminor version : 1
Input sensor was set to: Monocular

Loading ORB Vocabulary.
Vocabulary loaded!

Camera Parameters: 
- fx: 615.671
- fy: 615.962
- cx: 328.001
- cy: 241.31
- k1: 0
- k2: 0
- p1: 0
- p2: 0
- fps: 30
- bf: 1.77965e-43
- color order: RGB (ignored if grayscale)

ORB Extractor Parameters: 
- Number of Features: 2000
- Scale Levels: 8
- Scale Factor: 1.2
- Initial Fast Threshold: 20
- Minimum Fast Threshold: 7
Enable localization only: false
Map point vector is empty!
Map point vector is empty!
Map point vector is empty!
Map point vector is empty!
Map point vector is empty!
Map point vector is empty!
Map point vector is empty!

`
So, is there any specific solution to that, if so; could you explain? Furthermore, I did not get anything on rviz also. Almost all topics (generated after above runs) are empty such as pose etc.