appliedAI-Initiative/orb_slam_2_ros #4: map viewer

E-pep, 2019-02-08:
Hello, newbie here.

When i run the code the vocabulary loads, but it doesn't open a map viewer like in the original code. Am i doing something wrong?

lennarthaller, 2019-02-08:
Like stated in my readme: "For vizualization you can use RViz."

E-pep, 2019-02-08:
ah Thank you very much for fast answer

E-pep, 2019-02-10:
Hello, sorry to bother everyone but I have a problem. I am using the monolaunch and publishing to /camera/image_raw with usb_cam or with rosbag play and I see the images with image_view. When I visualize the /debug_image Image, I get No image received. I also don't see a point cloud. Can someone help?

lennarthaller, 2019-02-11:
As noted in the launch file for the mono slam, the algorithm expects the image at /camera/rgb/image_raw. I really suggest that you familiarize yourself with ros.

closed 2019-02-12 by lennarthaller