appliedAI-Initiative/orb_slam_2_ros #5: intel Euclid ROS1 Kinetic compiler problem

mockbot, 2019-02-15:
I've tried to compile orb_slam2_ros at my Intel Euclid Device (embed pc with buitin zr200) , but with the following result:

[ 87%] Building CXX object orb_slam_2_ros/CMakeFiles/orb_slam2_ros.dir/orb_slam2/src/System.cc.o
In file included from /usr/include/c++/5/thread:35:0,
                 from /intel/euclid/euclid_ws/src/orb_slam_2_ros/orb_slam2/include/System.h:26,
                 from /intel/euclid/euclid_ws/src/orb_slam_2_ros/orb_slam2/src/System.cc:23:
/usr/include/c++/5/bits/c++0x_warning.h:32:2: error: #error This file requires compiler and library support for the ISO C++ 2011 standard. This support must be enabled with the -std=c++11 or -std=gnu++11 compiler options.
 #error This file requires compiler and library support \
  ^
I've noticed, you changed something at compiler flags.

Compiling and running at another system (Intel NUC i5 + r200) works as expected.

Thanks and cheers

Chrimo

lennarthaller, 2019-02-15:
Thanks for the heads up. I just pushed something, let me know if your problem is fixed.

mockbot, 2019-02-17:
I've tested your changes with the following systems:

Intel Euclid, Ubuntu 16.04, ROS kinetic, ZR300 : compilation without warnings, node starts without warnings, OK

Inet NUC i5, Ubuntu 18.04, ROS melodic, R200 : compilation without warnings, node starts without warnings, OK

Perfect !!!
Thanks for your amazing fast response and fixes 

Cheers

Chrimo

lennarthaller, 2019-02-18:
Glad it works!
Closing this now.

closed 2019-02-18 by lennarthaller