appliedAI-Initiative/orb_slam_2_ros #7: Stereo camera?

abylikhsanov, 2019-02-22:
What do you think about this camera?
[LINK](https://realsense.intel.com/tracking-camera/?utm_source=realsense_website&utm_medium=banner&utm_campaign=launch_T265&utm_content=callout_index_learn-more_button
)
Do you think having such FoV and other its properties will help to extract orb points and construct a good map?

lennarthaller, 2019-02-25:
Hey,
interesting, looks pretty cool. A bigger FoV means more stable tracking in theory.
From glancing over the page I see that the device already runs some kind of SLAM. Would be cool to compare the performance.
I'd be interested in the results if you get them.

lennarthaller, 2019-04-21:
Closing this due to inactivity. Feel free to reopen if needed.

closed 2019-04-21 by lennarthaller

rakshith95, 2021-07-22:
Hello @abylikhsanov. Did you try to get the Intel T265 working with orb slam2?