appliedAI-Initiative/orb_slam_2_ros #9: Dense cloud compile issue

ChrisHughes, 2019-02-24:
Attempting to get the dense cloud branch to compile. In its current state, it is not compiling, and I have gone ahead and merged it with the master branch with the same result.

```
[ 95%] Linking CXX executable devel/lib/orb_slam2_ros/orb_slam2_ros_mono
orb_slam2/lib/liborb_slam2_ros.so: undefined reference to `TaskQueue::TaskQueue<boost::shared_ptr<pcl::PointCloud<pcl::PointXYZRGBA> >, boost::shared_ptr<pcl::PointCloud<pcl::PointXYZRGBA> >, boost::shared_ptr<pcl::PointCloud<pcl::PointXYZRGBA> >, cv::Mat>::~TaskQueue()'
orb_slam2/lib/liborb_slam2_ros.so: undefined reference to `TaskQueue::TaskQueue<boost::shared_ptr<pcl::PointCloud<pcl::PointXYZRGBA> >, boost::shared_ptr<pcl::PointCloud<pcl::PointXYZRGBA> >, boost::shared_ptr<pcl::PointCloud<pcl::PointXYZRGBA> >, cv::Mat>::NumJobsCurrentlyRunning()'
orb_slam2/lib/liborb_slam2_ros.so: undefined reference to `TaskQueue::TaskQueue<boost::shared_ptr<pcl::PointCloud<pcl::PointXYZRGBA> >, boost::shared_ptr<pcl::PointCloud<pcl::PointXYZRGBA> >, boost::shared_ptr<pcl::PointCloud<pcl::PointXYZRGBA> >, cv::Mat>::TaskQueue(unsigned int)'
collect2: error: ld returned 1 exit status
CMakeFiles/orb_slam2_ros_mono.dir/build.make:384: recipe for target 'devel/lib/orb_slam2_ros/orb_slam2_ros_mono' failed
make[2]: *** [devel/lib/orb_slam2_ros/orb_slam2_ros_mono] Error 1
CMakeFiles/Makefile2:106: recipe for target 'CMakeFiles/orb_slam2_ros_mono.dir/all' failed
make[1]: *** [CMakeFiles/orb_slam2_ros_mono.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 97%] Building CXX object CMakeFiles/orb_slam2_ros_stereo.dir/ros/src/Node.cc.o
[ 98%] Linking CXX executable devel/lib/orb_slam2_ros/orb_slam2_ros_rgbd
orb_slam2/lib/liborb_slam2_ros.so: undefined reference to `TaskQueue::TaskQueue<boost::shared_ptr<pcl::PointCloud<pcl::PointXYZRGBA> >, boost::shared_ptr<pcl::PointCloud<pcl::PointXYZRGBA> >, boost::shared_ptr<pcl::PointCloud<pcl::PointXYZRGBA> >, cv::Mat>::~TaskQueue()'
orb_slam2/lib/liborb_slam2_ros.so: undefined reference to `TaskQueue::TaskQueue<boost::shared_ptr<pcl::PointCloud<pcl::PointXYZRGBA> >, boost::shared_ptr<pcl::PointCloud<pcl::PointXYZRGBA> >, boost::shared_ptr<pcl::PointCloud<pcl::PointXYZRGBA> >, cv::Mat>::NumJobsCurrentlyRunning()'
orb_slam2/lib/liborb_slam2_ros.so: undefined reference to `TaskQueue::TaskQueue<boost::shared_ptr<pcl::PointCloud<pcl::PointXYZRGBA> >, boost::shared_ptr<pcl::PointCloud<pcl::PointXYZRGBA> >, boost::shared_ptr<pcl::PointCloud<pcl::PointXYZRGBA> >, cv::Mat>::TaskQueue(unsigned int)'
collect2: error: ld returned 1 exit status
CMakeFiles/orb_slam2_ros_rgbd.dir/build.make:384: recipe for target 'devel/lib/orb_slam2_ros/orb_slam2_ros_rgbd' failed
make[2]: *** [devel/lib/orb_slam2_ros/orb_slam2_ros_rgbd] Error 1
CMakeFiles/Makefile2:1830: recipe for target 'CMakeFiles/orb_slam2_ros_rgbd.dir/all' failed
make[1]: *** [CMakeFiles/orb_slam2_ros_rgbd.dir/all] Error 2
[100%] Linking CXX executable devel/lib/orb_slam2_ros/orb_slam2_ros_stereo
orb_slam2/lib/liborb_slam2_ros.so: undefined reference to `TaskQueue::TaskQueue<boost::shared_ptr<pcl::PointCloud<pcl::PointXYZRGBA> >, boost::shared_ptr<pcl::PointCloud<pcl::PointXYZRGBA> >, boost::shared_ptr<pcl::PointCloud<pcl::PointXYZRGBA> >, cv::Mat>::~TaskQueue()'
orb_slam2/lib/liborb_slam2_ros.so: undefined reference to `TaskQueue::TaskQueue<boost::shared_ptr<pcl::PointCloud<pcl::PointXYZRGBA> >, boost::shared_ptr<pcl::PointCloud<pcl::PointXYZRGBA> >, boost::shared_ptr<pcl::PointCloud<pcl::PointXYZRGBA> >, cv::Mat>::NumJobsCurrentlyRunning()'
orb_slam2/lib/liborb_slam2_ros.so: undefined reference to `TaskQueue::TaskQueue<boost::shared_ptr<pcl::PointCloud<pcl::PointXYZRGBA> >, boost::shared_ptr<pcl::PointCloud<pcl::PointXYZRGBA> >, boost::shared_ptr<pcl::PointCloud<pcl::PointXYZRGBA> >, cv::Mat>::TaskQueue(unsigned int)'
collect2: error: ld returned 1 exit status
CMakeFiles/orb_slam2_ros_stereo.dir/build.make:384: recipe for target 'devel/lib/orb_slam2_ros/orb_slam2_ros_stereo' failed
make[2]: *** [devel/lib/orb_slam2_ros/orb_slam2_ros_stereo] Error 1
CMakeFiles/Makefile2:68: recipe for target 'CMakeFiles/orb_slam2_ros_stereo.dir/all' failed
make[1]: *** [CMakeFiles/orb_slam2_ros_stereo.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
```

I have checked and rechecked the imports, package.xml and CMakeLists.txt to make sure the definition for 
```
TaskQueue::TaskQueue<DenseMap::PointCloudRGBD::Ptr, DenseMap::PointCloudRGBD::Ptr, DenseMap::PointCloudRGBD::Ptr, cv::Mat> *task_queue_;
```
is being included at the right spots.
At this point, I'm totally stuck and at a loss for how to proceed with this. I'm thinking possibly this is a linker bug relating to TaskQueue's Args... or that the definition needs to match a more generic type? 

I've forked the repo and pushed my working copy https://github.com/ChrisHughes/orb_slam_2_ros/commits/dense_map

Currently building this on Ubuntu 18.04, ROS melodic, C++ 7.3.0, cmake 3.10.2

lennarthaller, 2019-02-25:
Hey, unfortunately, as of now, the dense_map feature is not yet fully implemented. The plan still is to release it at some point but I can't say when that is gonna be.

closed 2019-02-28 by lennarthaller

StefanXR, 2020-07-12:
@lennarthaller Any update for the dense_map feature? I am getting the same Issue. Maybe in the last 1 1/2 years there is a update on this fork and the dense map feature is working? It would be great if so!