appliedAI-Initiative/orb_slam_2_ros #11: Use ardrone simulator camera?

20chix, 2019-02-26:
Hi

I was trying to use the camera of the ardrone from gazebo, but no luck. Do you think is possible?

lennarthaller, 2019-02-28:
From this statement I don't even know what and where the problem is.

20chix, 2019-02-28:
Sorry my bad, do you think is possible to use this project on a monocular camera? I'm planning on developing obstacle avoidance with slam. Tried to install it but didn't work

lennarthaller, 2019-02-28:
It is definitely possible to use this with a monocular camera ("for Monocular, Stereo and RGB-D" from the readme). Issue #8 is relevant for you then, you'd need to somehow calibrate your output to the right scale. For obstacle avoidance it is not very well suited since the resulting point cloud is sparse.
Not sure how well the algorithm works with camera input from a gazebo simulation.

closed 2019-02-28 by 20chix