appliedAI-Initiative/orb_slam_2_ros #12: Camera claibration

tnaderi, 2019-02-28:
Dear Lennart,

Thank you for providing us with your codes. 
I am trying to use http://wiki.ros.org/action/fullsearch/camera_calibration?action=fullsearch&context=180&value=linkto%3A%22camera_calibration%22 to calibrate a camera. The problem is the k3 is always 0 while in your examples the are not zero. In addition the format of yaml file is in matrix format. I checked open cv documents https://docs.opencv.org/2.4/doc/tutorials/calib3d/camera_calibration/camera_calibration.html and tried to match the parameters from matrix (open cv ) to your format. No luck. The point cloud is distorted. I would be grateful if you could help me in this regard.

Sincerely,
Taher Naderi

abylikhsanov, 2019-03-01:
What camera are you using? From his examples, all config files have k3 as zero

tnaderi, 2019-03-01:
I am using Logitech webcam C250. for example:

Camera.fx: 517.306408
Camera.fy: 516.469215
Camera.cx: 318.643040
Camera.cy: 255.313989

Camera.k1: 0.262383
Camera.k2: -0.953104
Camera.p1: -0.005358
Camera.p2: 0.002628
Camera.k3: 1.163314

is the setting from TUM1.yaml in example folders of ORB_SLAM2. While the settting from open cv is like:

image_width: 640
image_height: 480
camera_name: camera
camera_matrix:
  rows: 3
  cols: 3
  data: [688.133613, 0.000000, 346.628362, 0.000000, 687.421489, 265.646577, 0.000000, 0.000000, 1.000000]
distortion_model: plumb_bob
distortion_coefficients:
  rows: 1
  cols: 5
  data: [0.113606, -0.168358, 0.011983, 0.000844, 0.000000]
rectification_matrix:
  rows: 3
  cols: 3
  data: [1.000000, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000, 0.000000, 0.000000, 1.000000]
projection_matrix:
  rows: 3
  cols: 4
  data: [705.848022, 0.000000, 346.894846, 0.000000, 0.000000, 699.617432, 269.984300, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000]

What is the difference between the first matrix and the last one? In the example from calibration package the numbers from camera matrix and projection matrix are the same. Mine are a little different.

abylikhsanov, 2019-03-01:
What you need from your output are camera matrix and distortion coefficients.
From your camera matrix, positions 0 and 4 are fx and fy, positions 2 and 5 are cx and cy.
From your distortion matrix the coefficients, positions 0,1 and 4, so from your data k3 is zero and why you won't simply use it as 0?

tnaderi, 2019-03-01:
I did exactly what you said. Still the map from slam is distorted. 
[image]

tnaderi, 2019-03-01:
in TUM1.yaml and TUM2.yaml the k3 is not zero

tnaderi, 2019-03-01:
%YAML:1.0

# Camera calibration and distortion parameters (OpenCV)
# Camera calibration parameters (OpenCV)
Camera.fx: 688.133613
Camera.fy: 687.421489
Camera.cx: 346.628362
Camera.cy: 265.646577

# Camera distortion paremeters (OpenCV) --
Camera.k1: 0.113606
Camera.k2: -0.168358
Camera.p1: 0.011983
Camera.p2: 0.000844
Camera.k3: 0.0

Camera.width: 640
Camera.height: 480

# Camera frames per second
Camera.fps: 30.0

# Color order of the images (0: BGR, 1: RGB. It is ignored if images are grayscale)
Camera.RGB: 1

#--------------------------------------------------------------------------------------------
# ORB Parameters
#--------------------------------------------------------------------------------------------

# ORB Extractor: Number of features per image
ORBextractor.nFeatures: 2000

# ORB Extractor: Scale factor between levels in the scale pyramid
ORBextractor.scaleFactor: 1.2

# ORB Extractor: Number of levels in the scale pyramid
ORBextractor.nLevels: 8

# ORB Extractor: Fast threshold
# Image is divided in a grid. At each cell FAST are extracted imposing a minimum response.
# Firstly we impose iniThFAST. If no corners are detected we impose a lower value minThFAST
# You can lower these values if your images have low contrast
ORBextractor.iniThFAST: 20
ORBextractor.minThFAST: 7

lennarthaller, 2019-03-04:
That looks like the scale of this point cloud is just very small. Can you visualize your position in relation to the point cloud and see if that matches up?

tnaderi, 2019-03-04:
Yes it works actually. But it is not very precise. In addition, because it does not have lots of points on area with low frequency content like a wall I do not know how to use it for navigation. Where can I find the data of point clouds and how can I overcome that problem? I have a video stream (pipe file which comes from a drone) and I do not know how to publish that video as well. Sorry I am asking too many questions. Thank you for your response and time.

lennarthaller, 2019-03-04:
>  In addition, because it does not have lots of points on area with low frequency content like a wall I do not know how to use it for navigation

Im not sure I understand what you mean.

>  Where can I find the data of point clouds

There is no other way then subscribing to the point cloud.

> I have a video stream (pipe file which comes from a drone) and I do not know how to publish that video as well.

I think what you are looking for is rosbag (saving rostopics over time into a file)

lennarthaller, 2019-03-04:
> I did exactly what you said. Still the map from slam is distorted.
> [image]

Again, to me that doesn't look like the calibration is wrong but like the map is out of scale. So how does the change of position look in relation to the point cloud?

tnaderi, 2019-03-04:
I tried to change the position, it is giving me a fair point cloud, far better than the picture, which is good, not as good as the rgbd slam which was by far more precise. It is not really good for navigation, it does not see glass and it does not see area of images with low frequency content. Thank you.

tnaderi, 2019-03-04:
how should I change the scale of map?

abylikhsanov, 2019-03-04:
First of all, why are you expecting to get the same results from a cheap mono camera with rolling shutters as from the depth camera? I do not think you can get viable results from that camera.
Second of all, of course, you won't be able to detect the glass, it is a camera. There are no features on a glass.

It seems to me that you do not really understand what are you doing in this specific case and it is hard to give you the answer as I do not understand what exactly do you want.

tnaderi, 2019-03-04:
I am trying to do indoor navigation. I should use better camera. Is there anyway to detect glass using passive camera?

tnaderi, 2019-03-04:
I really appreciate that I have access to these codes. The algorithm is very intelligent and interesting. I am familiar with optical follow and why we need to change the position of camera. I am trying to understand better. Thank you.

tnaderi, 2019-03-04:
Feature detectors usually are able to detect high frequency content of image, right? So does SLAM, right?

lennarthaller, 2019-03-05:
> It seems to me that you do not really understand what you are doing at this specific case and it is hard to give you the answer as I do not understand what exactly do you want.

I agree with that. Please familiarize yourself with the concepts of what you are doing.

tnaderi, 2019-03-05:
Could you please answer this question: Feature detectors usually are able to detect high frequency content of image, right? So does SLAM, right?

lennarthaller, 2019-03-05:
Sorry, I don't know what you mean by "high frequency content". The features used for the slam are [ORB](http://www.willowgarage.com/sites/default/files/orb_final.pdf) features.

tnaderi, 2019-03-05:
Yes you are right. That is ORB. Look at the picture below please. The 
[image]. It is able to see the high frequency of image only, is it right? when we do not have texture there it is low frequency area because we do not have change in the image intensity. Right?

tnaderi, 2019-03-05:
We do not have corners, so it is low frequency area.

lennarthaller, 2019-03-05:
Ok, now I understand what you mean by frequency.
Yes, the SLAM only works if you have a lot of features to detect so a camera with a high FoV is potentially better. See #7 for that.

tnaderi, 2019-03-05:
Thank you!
In #7 you have mentioned "A bigger FoV means more stable tracking in theory." which makes a lot of sense. Thank you for your solid answer. It is clarifying. Sorry if I was not able to elaborate my question from the beginning. Thank you again. Have a nice day.

lennarthaller, 2019-03-08:
Perfect, so I'm going to close this now.

closed 2019-03-08 by lennarthaller