appliedAI-Initiative/orb_slam_2_ros #13: Wrong TF between MAP and Odom

abylikhsanov, 2019-03-01:
From my understanding, Node.cc provides a TF between camera and Map. I have changed camera to odom in order to not break my tree but seems that the transformation is wrong. This is the screenshot:

[image]

As you can see, the rotation is being changed in yaw and roll. 
I do not want to use the position approximation by openCV but I would like it to fuse it together with my odometry. 

Can you tell me if that suppose to happen?

lennarthaller, 2019-03-01:
Hey, 
thank you for reporting this.

> From my understanding, Node.cc provides a TF between camera and Ma
This is correct, that is the expected behavior. 

> As you can see, the rotation is being changed in yaw and roll.
Very strange indeed.

> Can you tell me if that suppose to happen?
This is not supposed to happen. I have never checked against the odom frame. I am going to look into this at some point (maybe in a week or two) but am very busy right now with other things.

If you want to implement a fix I'd be happy to get a pull request.

saoto28, 2019-03-08:
Hi, I have similar issue.
In my case, I just run the original launch files with realsense D435.
$ roslaunch realsense2_camera rs_rgbd.launch
$ roslaunch orb_slam2_ros orb_slam2_d435_rgbd.launch

This is the screenshot in which the camera_link is fixed frame. I can see the point cloud of the map above the camera as same as abylikhsanov's.
[image]
Let me know if you have any suggestion.

lennarthaller, 2019-03-08:
Also just in the Odom frame?
This needs to be fixed but I cant say at the moment when I get around to do it.
The transformations are calculated in the file ros/src/Node.cc if it's a pressing issue for now and you feel comfortable you can try to fix it.

lennarthaller, 2019-03-12:
[image]
Here is a screenshot of my tf tree which doesnt have odom. Can you show me how your tf tree looks like?
[image]
This is how the point cloud looks with this tf.

saoto28, 2019-03-13:
Thank you for the response.
In my case, no Odom frame is created, the tree is no problem, and the point cloud of map is created.
The only problem is just tf between map and camera_link. 
It may be different issue from abylikhsanov's.
I will make new topic for my issue.

lennarthaller, 2019-03-18:
Let me know if #15 which I just merged fixed your issue.

lennarthaller, 2019-04-21:
Closing this due to inactivity. Feel free to reopen if needed.

closed 2019-04-21 by lennarthaller