appliedAI-Initiative/orb_slam_2_ros #14: Wrong TF between map and camera_link

saoto28, 2019-03-13:
(moved from "Wrong TF between MAP and Odom #13")
It looks wrong tf between map and camera_link.

I just run the original launch files with realsense D435.
$ roslaunch realsense2_camera rs_rgbd.launch
$ roslaunch orb_slam2_ros orb_slam2_d435_rgbd.launch

This is the screenshot in which the camera_link is fixed frame. I can see the point cloud of the map above the camera, initially.
[image]

This is the tf tree.
[image]

I can make the point cloud map like this.
[image]
Here, I am using TUM1.yaml instead of RealSenseD435RGBD.yaml. Because of it was not stable with that.

This is initial output of 'rostopic echo tf'. I can see a twisted rotation.

transforms: 
  
    header: 
      seq: 0
      stamp: 
        secs: 1552466541
        nsecs: 638240644
      frame_id: "map"
    child_frame_id: "camera_link"
    transform: 
      translation: 
        x: -5.157461419e-05
        y: -3.99771925004e-05
        z: 5.12675360369e-05
      rotation: 
        x: -0.500074472098
        y: 0.500079234907
        z: -0.499891153489
        w: 0.499955105592

saoto28, 2019-03-14:
I may solve this issue.
I think we need to transform coordinates before making inverse matrix.
(Transform orb_slam coordinates system to Ros coordinate system at camera coordinates.)
After making inverse matrix, transform coordinates again at map coordinates.

It looks no problem with this code below, in Node.cc.
```
tf::Transform Node::TransformFromMat (cv::Mat position_mat) {
  cv::Mat rotation(3,3,CV_32F);
  cv::Mat translation(3,1,CV_32F);

  rotation = position_mat.rowRange(0,3).colRange(0,3);
  translation = position_mat.rowRange(0,3).col(3);

  tf::Matrix3x3 tf_camera_rotation (rotation.at<float> (0,0), rotation.at<float> (0,1), rotation.at<float> (0,2),
                                    rotation.at<float> (1,0), rotation.at<float> (1,1), rotation.at<float> (1,2),
                                    rotation.at<float> (2,0), rotation.at<float> (2,1), rotation.at<float> (2,2)
                                   );

  tf::Vector3 tf_camera_translation (translation.at<float> (0), translation.at<float> (1), translation.at<float> (2));

  //Coordinate transformation matrix : same as invYZ * Rz * Rx
  const tf::Matrix3x3 matTrans (0, 0, 1,
                               -1, 0, 0,
                                0,-1, 0);
                                
  //Convert from orb_slam coordinates system to ros coordinates system
  tf_camera_rotation = matTrans*tf_camera_rotation;
  tf_camera_translation = matTrans*tf_camera_translation;

  //Inverse matrix
  tf_camera_rotation = tf_camera_rotation.transpose();
  tf_camera_translation = -(tf_camera_rotation*tf_camera_translation);

  //Convert from orb_slam coordinates system to ros coordinates system
  tf_camera_rotation = matTrans*tf_camera_rotation;
  tf_camera_translation = matTrans*tf_camera_translation;

  return tf::Transform (tf_camera_rotation, tf_camera_translation);
}
```
(Next, I am investigating that the treatment of the distance of orb_slam is wrong.)

tiagojdias, 2019-03-15:
The fix proposed by @saoto28 really works.
I have just a question regarding your operations. 
Just posting a small recap to see if i am correct:
- Basically orb gives you the transformation from the  world to camera coordinates (extrinsic parameters R and t);
- Normally the linear movement is performed over x-y plane but for cameras in ROS it's in z-x plane;
- After you perform the inverse, which means that you are getting the camera to world transformation (camera pose);

Can you give some hint about the transformation after inverting the matrix?

saoto28, 2019-03-15:
Here is my understanding. Let me know if there is misunderstanding.
In the orb coordinate system, a flat plane is x-z and top is -y.
In the ros coordinate system, a flat plane is x-y and top is z.
"position_mat" which is output from orb looks the map position in the camera coordinates  wihich is the orb coordinate system.
After transforming coordinate, the matrix shows the map position in the camera coordinates which is the ros coordinate system.
After inversing, the matrix shows the camera position in the map coordinates which is the orb coordinate system.
After transforming coordinate, the matrix shows the camera position in the map coordinates which is the ros coordinate system.

tiagojdias, 2019-03-15:
Thank you for your feedback (it also makes sense to me).

saoto28, 2019-03-16:
I can make the point cloud of map ,using the RGD-D with intel D435.
But, the distance looks different between the map and camera view(ex: /camera/depth_registered/points) on rviz. Why?
[image]

lennarthaller, 2019-03-16:
Very cool, if the fix works please submit a pull request!

The point cloud in the picture looks out of scale, is your DepthMapFactor in your config right? The default value for the D435 is 1000 but can be changed in the realsense-viewer.

saoto28, 2019-03-17:
Thank you about "DepthMapFactor".
I was using wrong yaml file in which the DepthMapFactor was 5000 . When I use RealSenseD435RGBD.yaml, distance issue is gone.
(I changed ThDepth to 40 in yaml file, then the slam process became staible.)

I will try to submit a pull request.

saoto28, 2019-03-17:
I submitted the pull request #15.
It is first time submit. Let me know, if there is something strange.

lennarthaller, 2019-03-17:
Thank you, gonna test it tomorrow.

lennarthaller, 2019-03-18:
#15 is merged, your issue should be fixed now right?

saoto28, 2019-03-18:
Yes.

closed 2019-03-18 by saoto28