appliedAI-Initiative/orb_slam_2_ros #16: Zed camera

arafatsh, 2019-03-18:
I am using a zed camera, I made a .yaml file with my configurations and a launch file that corresponds to that file, when running it just stucks on this screen!
Any suggestions?   

[image]

lennarthaller, 2019-03-18:
Seems to me like the SLAM is running but doesnt get any input. Do your topics match? Do you get the debug image from the SLAM in rviz?

arafatsh, 2019-03-19:
Thanks for your respond, I think you are right.
This is my launch file 

[image]

It should subscribe to /zed/left/image_rect_color and /zed/right/image_rect_color which is published by zed wrapper zed.start 

but still my launch file is not subscribing to those nodes but subscribing to others 

[image]

in Rviz there is no image 

Do you know how to fix that!?
Thanks in advance

arafatsh, 2019-03-19:
Thank you for your respond, I had something wrong in the launch file.
Now its working.

lennarthaller, 2019-03-19:
Yes, your topics were swapped in your remap. Please don't delete your messages and familiarize yourself with ros.

arafatsh, 2019-03-19:
Exactly. 
you can close the issue.
thanks for your time

arafatsh, 2019-03-20:
One more question, I can view the point cloud in RViz can I view in another program? or is there a way to download Orbslam2 : Mapviewer?

lennarthaller, 2019-03-22:
No you can only use RViz for visualization.  Is there something missing you'd need?

arafatsh, 2019-03-22:
Can the published pointcloud give us a RGB 3D map?

lennarthaller, 2019-03-27:
No, a dense map is not published at this point.

closed 2019-03-28 by lennarthaller