appliedAI-Initiative/orb_slam_2_ros #19: Save an output map from OrbSlam2 and download the map to a Jetson Nano for localization

mandelyoni, 2019-04-20:
Dear Lennart,

We would like to save an output map from OrbSlam2 (after mapping a room with a laptop), and then download the map to a Jetson Nano, which will only run localization in the same room.
Any suggestions how to carry this out?

Thanks,  Yoni.

lennarthaller, 2019-04-21:
Right now, this feature is not supported but definitely something I plan on doing sometime in the future. If you want to work on this take a look at [this PR](https://github.com/raulmur/ORB_SLAM2/pull/381/commits/cb63a7cb2c5f8a1bc4f3efea7c9ee823758600a9) from the original implementation where someone worked on serializing the existing data structures.

Best,
Lennart

lennarthaller, 2019-05-06:
Closing due to inactivity. Feel free to reopen if needed.

r7vme, 2019-06-23:
Hello, just in case anyone wanted to save few hours. Here is up-to-date (Jun 23, 2019) map saver from [mentioned PR](https://github.com/raulmur/ORB_SLAM2/pull/381/commits/cb63a7cb2c5f8a1bc4f3efea7c9ee823758600a9). Tested with mono node, but others should also work.

```
git remote add map_saver https://github.com/r7vme/orb_slam_2_ros.git
git fetch map_saver
git cherry-pick ac24ff82ec6fddbf72f499421c8bc514bddfffc9
```

Few configs are changed:
- in TUM1.yaml added `Map.mapfile: map.bin`. By default ROS will save to `~/.ros/`.
- new ROS param `save_map`. Should be set to true, if want to save a map.

P.S.: I see branch [save-load](https://github.com/appliedAI-Initiative/orb_slam_2_ros/tree/save-load), but w/o any new code.

lennarthaller, 2019-06-24:
Very cool! If you create a PR I'm happy to test and merge it!

r7vme, 2019-06-25:
Hey @lennarthaller here is the PR https://github.com/appliedAI-Initiative/orb_slam_2_ros/pull/32

lennarthaller, 2019-06-25:
Awesome, perfect!

closed 2019-06-25 by lennarthaller