appliedAI-Initiative/orb_slam_2_ros #20: CMakeLists.txt correction - DBoW2 undefined reference

eMrazSVK, 2019-04-30:
Hi,

I use your orbslam2 ROS wrapper and appreciate your work. I am not sure, if I am the only one, but when I want to successfully run catkin_make to build this package I need to adjust CmakeLists.txt a bit.  

At lines: 

https://github.com/appliedAI-Initiative/orb_slam_2_ros/blob/9b8661e692f0911285b4c9131e4ad62c6b68c120/CMakeLists.txt#L114

https://github.com/appliedAI-Initiative/orb_slam_2_ros/blob/9b8661e692f0911285b4c9131e4ad62c6b68c120/CMakeLists.txt#L124

https://github.com/appliedAI-Initiative/orb_slam_2_ros/blob/9b8661e692f0911285b4c9131e4ad62c6b68c120/CMakeLists.txt#L134

I need to add `${LIBS_ORBSLAM}` to the target libraries to make it run properly. Otherwise I would get errors with undefined reference to DBoW2. So the commands for linking libraries for mono, stereo and rgbd look like this:

`target_link_libraries(${PROJECT_NAME}_mono
${LIBS_ROS} ${LIBS_ORBSLAM}
)`

`target_link_libraries(${PROJECT_NAME}_stereo
${LIBS_ROS} ${LIBS_ORBSLAM}
)`

`target_link_libraries(${PROJECT_NAME}_rgbd
${LIBS_ROS} ${LIBS_ORBSLAM}
)
`

That's one adjustment. Maybe you could ask other people, if that's an issue for them and change CMakeLists.txt here in your repo. Maybe it is just specific issue and it will help someone in the future, so it can be seen here. 

Keep up the good work you're doing!

lennarthaller, 2019-04-30:
Thanks for reporting this. I don't encounter this issue, but I'll investigate. Do you also get this error when using catkin build?

Best,
Lennart

eMrazSVK, 2019-05-01:
Oh yeah, that might be the problem. I used only catkin_make, because my workspace was previously built with it and I could not do it with catkin build. Maybe you could just add this to readme somewhere for catkin_make users.

lennarthaller, 2019-05-03:
Ok, perfect, so the solution is to apply your changes or use catkin build.
Going to close this then. Feel free to reopen if needed.

closed 2019-05-03 by lennarthaller