appliedAI-Initiative/orb_slam_2_ros #21: camera_calibration

guoxianzhen, 2019-05-03:
Hello,
  I am trying to use orb_slam_2_ros. But I only do the simulation in gazebo.  So I need to calibrate the simulated camera. So I want to ask if the 'camera_calibration' node can be used for a simulated camera? Or is there any other nodes for me to use for calibrating my camera.

lennarthaller, 2019-05-03:
I'm not familiar with simulated gazebo cameras but in theory you wouldn't need to run a calibration because the simulated cam just behaves like you specify so I'd imagine you need to provide cam intrinsics?
[This issue](https://github.com/appliedAI-Initiative/orb_slam_2_ros/issues/11) goes into the same direction.

lennarthaller, 2019-05-06:
Closing due to inactivity.

closed 2019-05-06 by lennarthaller