appliedAI-Initiative/orb_slam_2_ros #22: how to remap

guoxianzhen, 2019-05-06:
Hello,
  I am trying to use orb_slam2_ros. I would like to use orb_slam2_rgbd.  My camera give rgb image and depth image on topics "rgb/image_raw" and "depth/image_raw" respectively. But i am confusing about how to change the topic. Do i need to replace "/camera/rgb/image_rect_color" with "/rgb/image_raw" or replace "/camera/rgb/image_raw" with "/rgb/image_raw"?

 <remap from="/camera/rgb/image_raw" to="/camera/rgb/image_rect_color" />
<remap from="/camera/depth_registered/image_raw" to="/camera/depth_registered/sw_registered/image_rect" />

guoxianzhen, 2019-05-06:
I have changed the input image topic, But the orb_slam2_ros node didn't subscribe with the input image topic. Do you think what the problem is?

lennarthaller, 2019-05-06:
Seems like you have the same [problem](https://github.com/appliedAI-Initiative/orb_slam_2_ros/issues/16).

closed 2019-05-07 by lennarthaller