appliedAI-Initiative/orb_slam_2_ros #24: Creation of a listener for tf from map to camera_link

tommi95, 2019-05-08:
Hi,
I'm trying to use the transformation in order to know the camera pose in the map frame. Apparently this doesn't work, because the only existing frame is the camera_link frame with parent map, while no map frame without any parents exists  as expected. Is something known or I'm getting something wrong? Thanks
Toto

lennarthaller, 2019-05-08:
Hey,
thanks for raising this. Is publishing a map frame without parent a convention I'm not aware of? Or do I misunderstand your question?
The pointcloud has (as default) the map frame, so the relation from pose to pointcloud should work.
Best,
Lennart

tommi95, 2019-05-09:
Hi,
Thank you for your fast reply!! Probably I'm getting something wrong because I'm pretty new to this topic, so I apologize in advance.
In this case, after the initialization of a listener for the "/tf" in which the node ORB _SLAM2 is broadcasting, I try to check if the map frame exist using:

'''
listener.frameExists("map")

listener.waitForTransform('map', 'camera_link', now, rospy.Duration(0.05))

'''

Both functions always return false, no matter what. Should I use instead a different frame id for the transformation?
Thank very much!
Regards,
Tommaso

lennarthaller, 2019-05-13:
See this [comment](https://github.com/appliedAI-Initiative/orb_slam_2_ros/issues/13#issuecomment-472080897) I made in an other topic. Does your tf tree look similar?

lennarthaller, 2019-05-27:
Closing due to inactivity. Feel free to reopen if needed.

closed 2019-05-27 by lennarthaller