appliedAI-Initiative/orb_slam_2_ros #25: stereo rectification zed camera

TZVIh, 2019-05-12:
Hi,
I saw in the config file of orb slam some parameters for stereo rectification.
I want to use zed camera using this wrapper https://github.com/willdzeng/zed_cpu_ros and using only the images (right and left).
Do I need to rectify the images before publish to orbslam or it will rectify if I will give the stereo parameters?
Thanks!

lennarthaller, 2019-05-13:
If you provide the rectification parameters in your settings file the package will rectify the images for you.

TZVIh, 2019-05-22:
Hi,
Thanks for your answer.
I am sorry for my misunderstanding.
I created this yaml file.

> %YAML:1.0

#--------------------------------------------------------------------------------------------
# Camera Parameters. Adjust them!
#--------------------------------------------------------------------------------------------

# Camera calibration and distortion parameters (OpenCV) 
Camera.fx: 349.4
Camera.fy: 349.4
Camera.cx: 321.261
Camera.cy: 199.641

Camera.k1: -0.174077
Camera.k2: 0.0275047
Camera.p1: 0.0
Camera.p2: 0.0

Camera.width: 672
Camera.height: 376

# Camera frames per second 
Camera.fps: 15.0

# stereo baseline times fx
Camera.bf: 41.928

# Color order of the images (0: BGR, 1: RGB. It is ignored if images are grayscale)
Camera.RGB: 0

# Close/Far threshold. Baseline times.
ThDepth: 50

#--------------------------------------------------------------------------------------------
# Stereo Rectification. Only if you need to pre-rectify the images.
# Camera.fx, .fy, etc must be the same as in LEFT.P
#--------------------------------------------------------------------------------------------
LEFT.height: 376
LEFT.width: 672
LEFT.D: !!opencv-matrix
   rows: 1
   cols: 5
   dt: d
   data: [-0.174077, 0.0275047, 0.0, 0.0, 0.0]
LEFT.K: !!opencv-matrix
   rows: 3
   cols: 3
   dt: d
   data: [349.4, 0.0, 321.261, 0.0, 349.4, 199.641, 0.0, 0.0, 1.0]
LEFT.R:  !!opencv-matrix
   rows: 3
   cols: 3
   dt: d
   data: [0.9999791914104391, 0.0023169884216307905, 0.006020657005533152, -0.0023639914150825948, 0.9999666946626834, 0.007811600986801987, -0.006002357096479877, -0.00782567121987749, 0.9999513641068972]
LEFT.P:  !!opencv-matrix
   rows: 3
   cols: 4
   dt: d
   data: [349.4, 0.0, 321.261, 0.0, 0.0, 349.4, 199.641, 0.0, 0.0, 0.0, 1.0, 0.0]

RIGHT.height: 376
RIGHT.width: 672
RIGHT.D: !!opencv-matrix
   rows: 1
   cols: 5
   dt: d
   data: [-0.171929, 0.026479, 0.0, 0.0, 0.0]
RIGHT.K: !!opencv-matrix
   rows: 3
   cols: 3
   dt: d
   data: [350.602, 0.0, 329.986, 0.0, 350.602, 188.943, 0.0, 0.0, 1.0]
RIGHT.R:  !!opencv-matrix
   rows: 3
   cols: 3
   dt: d
   data: [0.9999791914104391, 0.0023169884216307905, 0.006020657005533152, -0.0023639914150825948, 0.9999666946626834, 0.007811600986801987, -0.006002357096479877, -0.00782567121987749, 0.9999513641068972]

RIGHT.P:  !!opencv-matrix
   rows: 3
   cols: 4
   dt: d
   data: [350.602, 0.0, 329.986, -41.928, 0.0, 350.602, 188.943, 0.0, 0.0, 0.0, 1.0, 0.0]

#--------------------------------------------------------------------------------------------
# ORB Parameters
#--------------------------------------------------------------------------------------------

# ORB Extractor: Number of features per image
ORBextractor.nFeatures: 1300

# ORB Extractor: Scale factor between levels in the scale pyramid 	
ORBextractor.scaleFactor: 1.2

# ORB Extractor: Number of levels in the scale pyramid	
ORBextractor.nLevels: 8

# ORB Extractor: Fast threshold
# Image is divided in a grid. At each cell FAST are extracted imposing a minimum response.
# Firstly we impose iniThFAST. If no corners are detected we impose a lower value minThFAST
# You can lower these values if your images have low contrast			
ORBextractor.iniThFAST: 20
ORBextractor.minThFAST: 7

#--------------------------------------------------------------------------------------------
# Viewer Parameters
#--------------------------------------------------------------------------------------------
Viewer.KeyFrameSize: 0.05
Viewer.KeyFrameLineWidth: 1
Viewer.GraphLineWidth: 0.9
Viewer.PointSize:2
Viewer.CameraSize: 0.08
Viewer.CameraLineWidth: 3
Viewer.ViewpointX: 0
Viewer.ViewpointY: -0.7
Viewer.ViewpointZ: -1.8
Viewer.ViewpointF: 500

> 

I saw that the node subscribe to:

> Subscribed topics

    The mono node subscribes to /camera/image_raw for the input image.

    The RGBD node subscribes to /camera/rgb/image_raw for the RGB image and

    /camera/depth_registered/image_raw for the depth information.

    The stereo node subscribes to image_left/image_color_rect and

    image_right/image_color_rect for corresponding images.

> 

so where the code rectify the images?

lennarthaller, 2019-05-22:
I'm right now not sure what your problem is. Does the rectification not work?

TZVIh, 2019-05-26:
Hi,
I created new launch file (copy of the realsense):

[orb_slam2_zed_stereo.txt](https://github.com/appliedAI-Initiative/orb_slam_2_ros/files/3220201/orb_slam2_zed_stereo.txt)

as I said I am using zed camera using this wrapper https://github.com/willdzeng/zed_cpu_ros and using only the images (the images not rectified).
In the launch I see that I need to remap the topics of the rectify images but and I saw that orb slam code will rectify the images if the parameters are in the config file.
what is the topics that I have to write in this launch?

TomyYamy, 2019-05-27:
I'm also interested in to exploit this pkg with ZED.
I tried that while referring your setting .yaml file.
[ZED_stereo.yaml.txt](https://github.com/appliedAI-Initiative/orb_slam_2_ros/files/3222162/ZED_stereo.yaml.txt)
I slightly changed your launch file.
[orb_slam2_zed_stereo.launch.txt](https://github.com/appliedAI-Initiative/orb_slam_2_ros/files/3222176/orb_slam2_zed_stereo.launch.txt)

It seems to start working while launching ZED from zed_cpu_ros pkg in my ubuntu 16.04 ROS kinetic environment.
[image]
Actually I'm not sure how to check it works correctly.

Thank you so much your submit. Actually this .yaml setting is complitley out of my depth. I could not try it without your submit.

TZVIh, 2019-05-27:
Hi @TomyYamy ,
where you got the "/zed/left/image_rect_color" topic?
how you created the rectified image?

TZVIh, 2019-05-27:
I found some solution. using the image_proc launch of zed cpu and recalibrate again the zed camera.

closed 2019-05-27 by TZVIh

NCHUPJ, 2020-05-16:
> Hi,
> I saw in the config file of orb slam some parameters for stereo rectification.
> I want to use zed camera using this wrapper https://github.com/willdzeng/zed_cpu_ros and using only the images [...]

Did you run it successfully?  Can you tell me how to set the parameters?