appliedAI-Initiative/orb_slam_2_ros #26: Orbslam 2 Error code -11

tiagojdias, 2019-05-16:
Hello, i've working with this version of orbslam2 for ros usage for a little while.

I've reworked the node in order to extract an odometry, which, in general, i think is the most relevant result to take in terms of SLAM algorithms.

Currently i'm experiencing a crash on ros Kinetic which gives an "error code -11". I haven't found a solution for that but re-spawning seems to fix it. I've looking into CPU, memory consumption and every is normal (the node itself consumes one thread at most).

This is the output:

[orb_slam2_d435_rgbd-47] process has died [pid 10046, exit code -11, cmd /home/agv/ros_ws/devel/lib/orb_slam2_ros/orb_slam2_ros_rgbd /home/agv/ros_ws/src/tb_ros/orb_slam_2_ros/orb_slam2/Vocabulary/ORBvoc.txt /home/agv/ros_ws/src/tb_ros/orb_slam_2_ros/orb_slam2/config/RealSenseD435RGBD.yaml /camera/rgb/image_raw:=/camera_front/color/image_raw /camera/depth_registered/image_raw:=/camera_front/aligned_depth_to_color/image_raw __name:=orb_slam2_d435_rgbd __log:=/home/agv/.ros/log/6f4f172c-77f0-11e9-9b09-e0d55e8c1304/orb_slam2_d435_rgbd-47.log].
log file: /home/agv/.ros/log/6f4f172c-77f0-11e9-9b09-e0d55e8c1304/orb_slam2_d435_rgbd-47*.log

Help would be appreciated, and if more information is needed i will provide.

lennarthaller, 2019-05-17:
Hm, if you modified the code it is going to be hard to help you with no context.
What do you mean by odometry? Do you want to publish odom messages? Be aware that the resulting path might not be continuous if a loop closure occurs.

tiagojdias, 2019-05-19:
Yes you are completely right.

I am using the SLAM mode on since my environment is pretty huge. Sometimes i can easily lose track so i have added in the ROS wrapper a reset mode (using the one that was previously created by Raul). Thus, everytime i lose track i'm able to restart the current process and continue. 

Now, since i'm applying the reset, my absolute localization is not reliable. My work around was to get two consecutive measurements and derive instantaneous velocities from them. Whenever i get a reset i discard the first measurement and start everything again, to ensure consistency.

It is working pretty well, there is only that exit code problem.
Just wanted to know if it was already familiar to you or the community itself.

Thanks for the reply.

lennarthaller, 2019-05-20:
Ah ok, now I understand. That is a very interesting feature!
Regarding the Error code -11 - I haven't encountered that myself yet but I am very happy that you reported it. I will do some testing. Are you memory constrained? Maybe your map size grows too large? I'm just speculating at this point.

Best,

Lennart

closed 2019-05-29 by tiagojdias