appliedAI-Initiative/orb_slam_2_ros #29: Launch File Issue

mmohmpat, 2019-06-12:
Hello,

I am having trouble launching ROS with the associated launch file using the following command,
`roslaunch orb_slam_2_ros orb_slam2_r200_mono.launch`

This produces the following error,
`RLException: [orb_slam2_r200_mono.launch] is neither a launch file in package [orb_slam_2_ros] nor is [orb_slam_2_ros] a launch file name
The traceback for the exception was written to the log file
`
I have tried following the readme and have executed the following,
`source devel/setup.bash`

I have also tried following the advice given by http://answers.ros.org/question/143496/roslaunch-is-neither-a-launch-file-in-package-nor-is-a-launch-file-name/. The `rospack find orb_slam_2_ros` command returned that the package was not found. Giving the absolute path for the commands tried did not yield different results.

(I am using ROS Melodic.)

Any help or advice would be greatly appreciated.

Thanks

lennarthaller, 2019-06-12:
This doesnt seem to be an issue with the launch file but with your setup. Given that you compiled this package without errors and ran  `source devel/setup.bash` in your catkin workspace you should be able to launch it.

mmohmpat, 2019-06-14:
Thank you for the prompt response, however, do you have any advice on what the setup issue may be or how to fix it?

closed 2019-06-14 by mmohmpat

sidml, 2019-12-05:
This error happens when you built the new workspace  with catkin build and the other workspace was built using catkin_make.

I solved this by making the workspace using catkin_make.

This question is relevant. https://answers.ros.org/question/205976/sourcing-from-multiple-workspaces/