appliedAI-Initiative/orb_slam_2_ros #30: Cannot launch node of type []: can't locate node [] in package []

mmohmpat, 2019-06-14:
Hi,

When I try running `roslaunch orb_slam2_ros orb_slam2_r200_mono.launch` the following error appears 
`ERROR: cannot launch node of type [orb_slam2_ros/orb_slam2_ros_mono]: can't locate node [orb_slam2_ros_mono] in package [orb_slam2_ros]`

Any help troubleshooting the issue? If it an issue with setup then what in particular needs to be changed in the installation to solve this? 

Thanks

lennarthaller, 2019-06-17:
Duplicate of #29? I still believe this is an error with your installation and is not related to this package. But maybe I'm wrong. Anyone else experiencing this issue?

mmohmpat, 2019-06-18:
Do you know what error in installation would cause this? Mismatch in versions or missing packages or similar?

How could I fix this issue?

lennarthaller, 2019-06-18:
My guess would be that the ROS workspace variables are not set or set wrong. Try if you find something on Google related to this. This is a generic error and I am almost certain its not related to this package.

closed 2019-06-25 by lennarthaller