appliedAI-Initiative/orb_slam_2_ros #33: Unable to record Pose

tekrajchhetri, 2019-06-26:
Hi,
below are my published and subscribed topic. When I try recording the topic by playing the source bag file, it doesn't record all the topic like pose, mappoints. Nor does it save the result in *.txt file. Cab anyone @lennarthaller  help?

[image]

Thanks

lennarthaller, 2019-06-26:
That the saving of the trajectory to .txt doesn't work is a good point but a known problem.
Can you echo the pose? If so you should be able to record to bag.

tekrajchhetri, 2019-06-26:
It is really strange i was trying the same and was unable to do earlier, spent lot of time. But now after your message I retried again and its working and was able to record the pose. It can be closed. Thanks.

closed 2019-06-26 by tekrajchhetri