appliedAI-Initiative/orb_slam_2_ros #34: Service Save_Map is not available

spurohit01, 2019-07-26:
Hi, 
I was able to successfully launch orb_slam2_ros with my monocular camera (PiCam) after specifying my own calibration file. Unfortunately, when I wanted to save the map using rosservice call /orb_slam2_mono/save_map map.bin after launching, I received the error "Service [/orb_slam2_mono/save_map] is not available". When I ran "rosservice list" there were six different services for the /orb_slam2_mono/ node, but save_map wasn't one of them. Just wondering what step I am missing or what I am doing wrong. 

Thanks!

lennarthaller, 2019-07-26:
Hey,
did you source the catkin workspace in the terminal from which you are calling the service using
`source devel/setup.bash`?

spurohit01, 2019-07-26:
Thanks for getting back to me so quickly! I did source setup.bash in that terminal before calling the service but unfortunately I still got the error.

lennarthaller, 2019-07-26:
Are you running the most current version? The save-load feature was just released yesterday.

spurohit01, 2019-07-26:
I am, I cloned it yesterday and just did git pull right now to confirm but it says the only things changed were two lines in the READ-ME.

lennarthaller, 2019-07-26:
Strange. Can you launch the node with a launch file which ships with this repo just to check if this might be the reason? I doubt it though.
Otherwise I try to find some time on monday to look at it more closely.

spurohit01, 2019-07-26:
I tried it with the Intel Real Sense r200 launch files for both monocular and stereo but I got the same error when running the respective rosservice. It would be great if you could please look into it. Thanks!

lennarthaller, 2019-07-29:
The error seems to be specific to your setup. My guess is that the setup bash is not (correctly) sourced. Without more information I cannot reproduce the error.

spurohit01, 2019-07-29:
Ok, I will try again. By the way, do I need to specify a ROS_PACKAGE_PATH? 

Thanks.

lennarthaller, 2019-07-30:
Glad it works! 
No, if you subscribe to the debug_image from the orb_slam2 node in RViz you'll see the output.

Best,
Lennart

spurohit01, 2019-07-30:
Great! I can now see my camera feed. But is visual feature detection on the camera feed supported? I don't see any markers on the feed in RViz pointing out any features and the DepthCloud is empty even after specifying my own camera calibration values and calling the rosservice to save the map. My goal is to be able to localize the camera after mapping but I'm unsure how to do this. 
Once again thanks for all your help.

lennarthaller, 2019-07-30:
You are using mono, so this is relevant: https://github.com/appliedAI-Initiative/orb_slam_2_ros/issues/2

spurohit01, 2019-07-30:
Hi Lennart, thanks, I just read the thread. Unfortunately in my case I don't see the green lines or any feature detection after moving the camera around for a while as you had mentioned in that answer. In RViz I am able to see the camera feed under the "image" section, but the "camera" section says "no image recieved" even though I have specified the same topic for both in the dropdown (see first picture). Additionally, I get an warning in terminal as shown in the second picture. I am assuming the warning is related to the camera not having got the points yet as described in the thread, but I just want to confirm. 
I'd really appreciate any suggestions you might have. Thanks so much!
[image]
[image]

lennarthaller, 2019-07-31:
The error message you are getting is pretty self explanatory: The SLAM doesnt accept frames from your camera because the frame_id is empty. Depending on which cam and ROS node you use you can change that in the settings or need to hardcode that into the node. Default for the frame_id is "camera_link".

This really doesn't have anything to do with this package.

closed 2019-07-31 by lennarthaller