appliedAI-Initiative/orb_slam_2_ros #37: orb slam pose correction

abylikhsanov, 2019-08-13:
Question, I have a wheel odometry and due to its covariance it will eventually bring up errors. So I was thinking to use orb slam to correct the pose. What is the best practice for that? I was thinking to correct my odometry message from the wheel encoders every time the loop closure is being detected (from this code):
https://github.com/appliedAI-Initiative/orb_slam_2_ros/blob/master/orb_slam2/src/LoopClosing.cc

Is that a good way to correct the pose? Because feeding the orb slam visual odometry to the sensor fusion algorithms like EKF in robot_localization package is not a solution, as I cannot make it as an absolute position that will make the covariances to collapse.

lennarthaller, 2019-08-14:
> Because feeding the orb slam visual odometry to the sensor fusion algorithms like EKF in robot_localization package is not a solution, as I cannot make it as an absolute position that will make the covariances to collapse.

I don't think I understand what you mean here. What is your ultimate goal? Do you really want to improve your wheel odometry or do you want to get a good position estimate of your robot? 

If it is the latter I'd suggest leaving the odometry as it is and extending the pose publisher of orb_slam_2_ros to publish a [PoseWithCovarianceStamped](http://docs.ros.org/api/geometry_msgs/html/msg/PoseWithCovarianceStamped.html) which can be consumed by the robot_localization package.

abylikhsanov, 2019-08-14:
Yeah, the latter. Does the orb slam gives the covariancr indecies? Also, do
they collapse after loop closure?

I am also thinking to add wheel odometry whenever the track is lost, do you
think it is a good idea?

Regarding the orb slam, I have launched the file but getting logs that 0
matches found and the map vector is empty, but I am supplying the raw
images (not rectified) with camera info prameters. What could it be?

abylikhsanov, 2019-08-14:
Ok, I can see that it publishes the covariance. One question, where exactly does the Tracking thread store the position values? Because I am planning to store some wheel odometry values and use them whenever the visual will be lost.

Ador2, 2019-09-25:
Tcw is 4x4 transformation matrix (camera frame to world frame), first 3x3 elements are the rotation matrix and elements(rows(1,2,3), column(4)) are x,y,z translations respectively.

closed 2019-10-08 by lennarthaller