appliedAI-Initiative/orb_slam_2_ros #40: Astra camera

hoonkai, 2019-09-01:
Hi

I'm trying to extend support to Orbbec Astra Pro. I can successfully produce keypoints, but unfortunately point cloud cannot be shown:

[image]

The launch file is as follows:

```
<launch>
  <node name="orb_slam2_rgbd" pkg="orb_slam2_ros"
      type="orb_slam2_ros_rgbd" output="screen">

       <param name="publish_pointcloud" type="bool" value="true" />
       <param name="publish_pose" type="bool" value="true" />
       <param name="localize_only" type="bool" value="false" />
       <param name="reset_map" type="bool" value="false" />

       <!-- static parameters -->
       <param name="load_map" type="bool" value="false" />
       <param name="map_file" type="string" value="map.bin" />
       <param name="settings_file" type="string" value="$(find orb_slam2_ros)/orb_slam2/config/TUM1.yaml" />
       <param name="voc_file" type="string" value="$(find orb_slam2_ros)/orb_slam2/Vocabulary/ORBvoc.txt" />

       <param name="pointcloud_frame_id" type="string" value="map" />
       <param name="camera_frame_id" type="string" value="camera_link" />
       <param name="min_num_kf_in_map" type="int" value="5" />
  </node>
</launch>
```

Does anyone know why the point cloud cannot be shown even when the fixed frame is already set to `map`?

Thanks in advance.

hoonkai, 2019-09-03:
I should add that `rostopic echo` on `map_points` give all zeros.

Thanks

hoonkai, 2019-09-03:
I should add that `rostopic echo` on map_points give non-zero data.

Thanks

closed 2019-09-12 by hoonkai

cagrihakkoymaz, 2020-10-08:
I encountered the same problem.Could you suggest ans solution?