appliedAI-Initiative/orb_slam_2_ros #41: error installing orb_slam_2_ros using catkin_make with OpenCV 4.1.0

SalilVishnuKapur, 2019-09-18:
Issue :
I have compiled opencv 4.1.0 with cuda support. So want to build the ros dependent packages with custom opencv. The command after initilizing catkin_init_workspace is as follows. Any kind of help will be deeply appreciated
Command :
catkin_make -DCMAKE_BUILD_TYPE=Release -DOpenCV_DIR=/opt/include/opencv4/opencv2 -DPYTHON_EXECUTABLE=/usr/bin/python3

My fixes till yet :
I fixed the initial issues by editing the cmake file to have opencv version 4.1.0 and also /home/spiri/ros_opencv_dependables/src/orb_slam_2_ros/orb_slam2/include/ORBextractor.h to have #include <opencv2/opencv.hpp> instead of #include <opencv/cv.h> 

My system config :
Ubuntu 18.04
Opencv 4.1.0
gcc 7.4.0
catkin

Error Log :
[ 64%] Building CXX object orb_slam_2_ros/CMakeFiles/orb_slam2_ros.dir/orb_slam2/src/FrameDrawer.cc.o
[ 65%] Building CXX object orb_slam_2_ros/CMakeFiles/orb_slam2_ros.dir/orb_slam2/src/MapPoint.cc.o
In file included from /home/spiri/ros_opencv_dependables/src/orb_slam_2_ros/orb_slam2/src/Tracking.cc:34:0:
/home/spiri/ros_opencv_dependables/src/orb_slam_2_ros/orb_slam2/include/PnPsolver.h:96:15: error: ‘CvMat’ has not been declared
   void fill_M(CvMat * M, const int row, const double * alphas, const double u, const double v);
               ^~~~~
/home/spiri/ros_opencv_dependables/src/orb_slam_2_ros/orb_slam2/include/PnPsolver.h:102:34: error: ‘CvMat’ does not name a type
   void find_betas_approx_1(const CvMat * L_6x10, const CvMat * Rho, double * betas);
                                  ^~~~~
/home/spiri/ros_opencv_dependables/src/orb_slam_2_ros/orb_slam2/include/PnPsolver.h:102:56: error: ‘CvMat’ does not name a type
   void find_betas_approx_1(const CvMat * L_6x10, const CvMat * Rho, double * betas);
                                                        ^~~~~
/home/spiri/ros_opencv_dependables/src/orb_slam_2_ros/orb_slam2/include/PnPsolver.h:103:34: error: ‘CvMat’ does not name a type
   void find_betas_approx_2(const CvMat * L_6x10, const CvMat * Rho, double * betas);
                                  ^~~~~
/home/spiri/ros_opencv_dependables/src/orb_slam_2_ros/orb_slam2/include/PnPsolver.h:103:56: error: ‘CvMat’ does not name a type
   void find_betas_approx_2(const CvMat * L_6x10, const CvMat * Rho, double * betas);
                                                        ^~~~~
/home/spiri/ros_opencv_dependables/src/orb_slam_2_ros/orb_slam2/include/PnPsolver.h:104:34: error: ‘CvMat’ does not name a type
   void find_betas_approx_3(const CvMat * L_6x10, const CvMat * Rho, double * betas);
                                  ^~~~~
/home/spiri/ros_opencv_dependables/src/orb_slam_2_ros/orb_slam2/include/PnPsolver.h:104:56: error: ‘CvMat’ does not name a type
   void find_betas_approx_3(const CvMat * L_6x10, const CvMat * Rho, double * betas);
                                                        ^~~~~
/home/spiri/ros_opencv_dependables/src/orb_slam_2_ros/orb_slam2/include/PnPsolver.h:105:17: error: ‘CvMat’ has not been declared
   void qr_solve(CvMat * A, CvMat * b, CvMat * X);
                 ^~~~~
/home/spiri/ros_opencv_dependables/src/orb_slam_2_ros/orb_slam2/include/PnPsolver.h:105:28: error: ‘CvMat’ has not been declared
   void qr_solve(CvMat * A, CvMat * b, CvMat * X);
                            ^~~~~
/home/spiri/ros_opencv_dependables/src/orb_slam_2_ros/orb_slam2/include/PnPsolver.h:105:39: error: ‘CvMat’ has not been declared
   void qr_solve(CvMat * A, CvMat * b, CvMat * X);
                                       ^~~~~
/home/spiri/ros_opencv_dependables/src/orb_slam_2_ros/orb_slam2/include/PnPsolver.h:113:27: error: ‘CvMat’ does not name a type
   void gauss_newton(const CvMat * L_6x10, const CvMat * Rho, double current_betas[4]);
                           ^~~~~
/home/spiri/ros_opencv_dependables/src/orb_slam_2_ros/orb_slam2/include/PnPsolver.h:113:49: error: ‘CvMat’ does not name a type
   void gauss_newton(const CvMat * L_6x10, const CvMat * Rho, double current_betas[4]);
                                                 ^~~~~
/home/spiri/ros_opencv_dependables/src/orb_slam_2_ros/orb_slam2/include/PnPsolver.h:115:23: error: ‘CvMat’ has not been declared
         double cb[4], CvMat * A, CvMat * b);
                       ^~~~~
/home/spiri/ros_opencv_dependables/src/orb_slam_2_ros/orb_slam2/include/PnPsolver.h:115:34: error: ‘CvMat’ has not been declared
         double cb[4], CvMat * A, CvMat * b);
                                  ^~~~~
[ 65%] Building CXX object orb_slam_2_ros/CMakeFiles/orb_slam2_ros.dir/orb_slam2/src/KeyFrame.cc.o
/home/spiri/ros_opencv_dependables/src/orb_slam_2_ros/orb_slam2/src/Tracking.cc: In member function ‘cv::Mat ORB_SLAM2::Tracking::GrabImageStereo(const cv::Mat&, const cv::Mat&, const double&)’:
/home/spiri/ros_opencv_dependables/src/orb_slam_2_ros/orb_slam2/src/Tracking.cc:171:38: error: ‘CV_RGB2GRAY’ was not declared in this scope
             cvtColor(mImGray,mImGray,CV_RGB2GRAY);
                                      ^~~~~~~~~~~
/home/spiri/ros_opencv_dependables/src/orb_slam_2_ros/orb_slam2/src/Tracking.cc:171:38: note: suggested alternative: ‘CV_RGB’
             cvtColor(mImGray,mImGray,CV_RGB2GRAY);
                                      ^~~~~~~~~~~
                                      CV_RGB
/home/spiri/ros_opencv_dependables/src/orb_slam_2_ros/orb_slam2/src/Tracking.cc:176:38: error: ‘CV_BGR2GRAY’ was not declared in this scope
[... 582 lines cut ...]
                          ^~~~~
/home/spiri/ros_opencv_dependables/src/orb_slam_2_ros/orb_slam2/src/PnPsolver.cc:860:26: error: ‘CvMat’ was not declared in this scope
/home/spiri/ros_opencv_dependables/src/orb_slam_2_ros/orb_slam2/src/PnPsolver.cc:860:34: error: ‘A’ was not declared in this scope
 void PnPsolver::qr_solve(CvMat * A, CvMat * b, CvMat * X)
                                  ^
/home/spiri/ros_opencv_dependables/src/orb_slam_2_ros/orb_slam2/src/PnPsolver.cc:860:37: error: ‘CvMat’ was not declared in this scope
 void PnPsolver::qr_solve(CvMat * A, CvMat * b, CvMat * X)
                                     ^~~~~
/home/spiri/ros_opencv_dependables/src/orb_slam_2_ros/orb_slam2/src/PnPsolver.cc:860:45: error: ‘b’ was not declared in this scope
 void PnPsolver::qr_solve(CvMat * A, CvMat * b, CvMat * X)
                                             ^
/home/spiri/ros_opencv_dependables/src/orb_slam_2_ros/orb_slam2/src/PnPsolver.cc:860:48: error: ‘CvMat’ was not declared in this scope
 void PnPsolver::qr_solve(CvMat * A, CvMat * b, CvMat * X)
                                                ^~~~~
/home/spiri/ros_opencv_dependables/src/orb_slam_2_ros/orb_slam2/src/PnPsolver.cc:860:56: error: ‘X’ was not declared in this scope
 void PnPsolver::qr_solve(CvMat * A, CvMat * b, CvMat * X)
                                                        ^
orb_slam_2_ros/CMakeFiles/orb_slam2_ros.dir/build.make:206: recipe for target 'orb_slam_2_ros/CMakeFiles/orb_slam2_ros.dir/orb_slam2/src/FrameDrawer.cc.o' failed
make[2]: *** [orb_slam_2_ros/CMakeFiles/orb_slam2_ros.dir/orb_slam2/src/FrameDrawer.cc.o] Error 1
orb_slam_2_ros/CMakeFiles/orb_slam2_ros.dir/build.make:350: recipe for target 'orb_slam_2_ros/CMakeFiles/orb_slam2_ros.dir/orb_slam2/src/PnPsolver.cc.o' failed
make[2]: *** [orb_slam_2_ros/CMakeFiles/orb_slam2_ros.dir/orb_slam2/src/PnPsolver.cc.o] Error 1
CMakeFiles/Makefile2:9039: recipe for target 'orb_slam_2_ros/CMakeFiles/orb_slam2_ros.dir/all' failed
make[1]: *** [orb_slam_2_ros/CMakeFiles/orb_slam2_ros.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2

lennarthaller, 2020-04-02:
Were you able to fix the issue?

lennarthaller, 2020-04-10:
Let me know if the issues persists after #62

deadmeat17, 2020-08-23:
> Let me know if the issues persists after #62

Yes it is still there.

aaasdream, 2020-10-14:
Add this line
```
using namespace cv;
```
It's work to me.

aaasdream, 2020-10-14:
Add This:
```
#include <opencv2/imgproc/types_c.h>
#include <opencv2/opencv.hpp>
using namespace cv;
```
It's work to me.

stephennacion06, 2020-10-14:
> Add This:
> 
> ```
> #include <opencv2/imgproc/types_c.h>
> #include <opencv2/opencv.hpp>
> using namespace cv;
> ```
> 
> It's work to me.

What file did you add that lines

Goh-tsen, 2020-12-03:
> > Add This:
> > ```
> > #include <opencv2/imgproc/types_c.h>
> > #include <opencv2/opencv.hpp>
> > using namespace cv;
> > ```
> > 
> > 
> > It's work to me.
> 
> What file did you add that lines

At first, I added these lines to each file that error messages mentioned and that really worked. But I had to edit numbers of files after each unsuccessful compiling. Then I find all those files contain #include <System.h>, so I added these three lines in the ORB_SLAM2/include/System.h. However I don't know if it really worked because it seemed that I have added these lines to all the files that may cause problem before I change the System.h.

eladden, 2020-12-25:
> At first, I added these lines to each file that error messages mentioned and that really worked. But I had to edit numbers of files after each unsuccessful compiling. Then I find all those files [...]

I've added these lines to System.h, PnPSolver.h, Sim3Solver.h only and it seemed to work

xlla, 2021-08-30:
After add those header files, I encounter another errors:
`
src/PnPsolver.cc:388:17: error: ‘cvCreateMat’ was not declared in this scope
   CvMat * PW0 = cvCreateMat(number_of_correspondences, 3, CV_64F);
                 ^~~~~~~~~~~

`