appliedAI-Initiative/orb_slam_2_ros #43: catkin build error: undefined reference to DUtils and DBoW2

pedrobagatin, 2019-10-08:
```
/usr/bin/ld: warning: libopencv_core.so.3.4, needed by /home/pedro/catkin_ws/src/orb_slam_2_ros/orb_slam2/lib/liborb_slam2_ros.so, may conflict with libopencv_core.so.3.2
/usr/bin/ld: warning: libopencv_imgcodecs.so.3.2, needed by /opt/ros/melodic/lib/libcv_bridge.so, may conflict with libopencv_imgcodecs.so.3.4
/home/pedro/catkin_ws/src/orb_slam_2_ros/orb_slam2/lib/liborb_slam2_ros.so: undefined reference to `DUtils::Random::SeedRandOnce()'
/home/pedro/catkin_ws/src/orb_slam_2_ros/orb_slam2/lib/liborb_slam2_ros.so: undefined reference to `DUtils::Random::RandomInt(int, int)'
/home/pedro/catkin_ws/src/orb_slam_2_ros/orb_slam2/lib/liborb_slam2_ros.so: undefined reference to `DUtils::Random::SeedRandOnce(int)'
/home/pedro/catkin_ws/src/orb_slam_2_ros/orb_slam2/lib/liborb_slam2_ros.so: undefined reference to `DBoW2::FORB::toArray8U(cv::Mat const&, unsigned char*)'
/home/pedro/catkin_ws/src/orb_slam_2_ros/orb_slam2/lib/liborb_slam2_ros.so: undefined reference to `DBoW2::FORB::L'
/home/pedro/catkin_ws/src/orb_slam_2_ros/orb_slam2/lib/liborb_slam2_ros.so: undefined reference to `DBoW2::FORB::fromArray8U(cv::Mat&, unsigned char*)'
collect2: error: ld returned 1 exit status
make[2]: *** [/home/pedro/catkin_ws/devel/.private/orb_slam2_ros/lib/orb_slam2_ros/orb_slam2_ros_rgbd] Error 1
make[1]: *** [CMakeFiles/orb_slam2_ros_rgbd.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
/usr/bin/ld: warning: libopencv_core.so.3.4, needed by /home/pedro/catkin_ws/src/orb_slam_2_ros/orb_slam2/lib/liborb_slam2_ros.so, may conflict with libopencv_core.so.3.2
/usr/bin/ld: warning: libopencv_imgcodecs.so.3.2, needed by /opt/ros/melodic/lib/libcv_bridge.so, may conflict with libopencv_imgcodecs.so.3.4
/home/pedro/catkin_ws/src/orb_slam_2_ros/orb_slam2/lib/liborb_slam2_ros.so: undefined reference to `DUtils::Random::SeedRandOnce()'
/home/pedro/catkin_ws/src/orb_slam_2_ros/orb_slam2/lib/liborb_slam2_ros.so: undefined reference to `DUtils::Random::RandomInt(int, int)'
/home/pedro/catkin_ws/src/orb_slam_2_ros/orb_slam2/lib/liborb_slam2_ros.so: undefined reference to `DUtils::Random::SeedRandOnce(int)'
/home/pedro/catkin_ws/src/orb_slam_2_ros/orb_slam2/lib/liborb_slam2_ros.so: undefined reference to `DBoW2::FORB::toArray8U(cv::Mat const&, unsigned char*)'
/home/pedro/catkin_ws/src/orb_slam_2_ros/orb_slam2/lib/liborb_slam2_ros.so: undefined reference to `DBoW2::FORB::L'
/home/pedro/catkin_ws/src/orb_slam_2_ros/orb_slam2/lib/liborb_slam2_ros.so: undefined reference to `DBoW2::FORB::fromArray8U(cv::Mat&, unsigned char*)'
collect2: error: ld returned 1 exit status
make[2]: *** [/home/pedro/catkin_ws/devel/.private/orb_slam2_ros/lib/orb_slam2_ros/orb_slam2_ros_mono] Error 1
make[1]: *** [CMakeFiles/orb_slam2_ros_mono.dir/all] Error 2
/usr/bin/ld: warning: libopencv_core.so.3.4, needed by /home/pedro/catkin_ws/src/orb_slam_2_ros/orb_slam2/lib/liborb_slam2_ros.so, may conflict with libopencv_core.so.3.2
/usr/bin/ld: warning: libopencv_imgcodecs.so.3.2, needed by /opt/ros/melodic/lib/libcv_bridge.so, may conflict with libopencv_imgcodecs.so.3.4
/home/pedro/catkin_ws/src/orb_slam_2_ros/orb_slam2/lib/liborb_slam2_ros.so: undefined reference to `DUtils::Random::SeedRandOnce()'
/home/pedro/catkin_ws/src/orb_slam_2_ros/orb_slam2/lib/liborb_slam2_ros.so: undefined reference to `DUtils::Random::RandomInt(int, int)'
/home/pedro/catkin_ws/src/orb_slam_2_ros/orb_slam2/lib/liborb_slam2_ros.so: undefined reference to `DUtils::Random::SeedRandOnce(int)'
/home/pedro/catkin_ws/src/orb_slam_2_ros/orb_slam2/lib/liborb_slam2_ros.so: undefined reference to `DBoW2::FORB::toArray8U(cv::Mat const&, unsigned char*)'
/home/pedro/catkin_ws/src/orb_slam_2_ros/orb_slam2/lib/liborb_slam2_ros.so: undefined reference to `DBoW2::FORB::L'
/home/pedro/catkin_ws/src/orb_slam_2_ros/orb_slam2/lib/liborb_slam2_ros.so: undefined reference to `DBoW2::FORB::fromArray8U(cv::Mat&, unsigned char*)'
collect2: error: ld returned 1 exit status
make[2]: *** [/home/pedro/catkin_ws/devel/.private/orb_slam2_ros/lib/orb_slam2_ros/orb_slam2_ros_stereo] Error 1
make[1]: *** [CMakeFiles/orb_slam2_ros_stereo.dir/all] Error 2
make: *** [all] Error 2

```

Andrea485, 2019-10-17:
in CmakeList.txt add 

> ${PROJECT_SOURCE_DIR}/orb_slam2/Thirdparty/DBoW2/lib/libDBoW2.so

like this:

> set(LIBS_ROS
> ${PROJECT_SOURCE_DIR}/orb_slam2/lib/lib${PROJECT_NAME}.so
> ${PROJECT_SOURCE_DIR}/orb_slam2/Thirdparty/DBoW2/lib/libDBoW2.so
> ${OpenCV_LIBS}
> ${catkin_LIBRARIES}
> )

closed 2020-04-02 by lennarthaller

blackjacket996, 2023-04-20:
> 

I met the same question! But I use ros2 ,what should I do ？

spectra543, 2023-07-28:
> > 
> 
> I met the same question! But I use ros2 ,what should I do ？

hi did u solved the problem yet? if so how? i also use ros2

pedrobagatin, 2023-07-28:
> > > 
> > 
> > 
> > I met the same question! But I use ros2 ,what should I do ？
> 
> hi did u solved the problem yet? if so how? i also use ros2

I am not sure about ROS2, was using ROS at the time

spectra543, 2023-07-28:
> > > > 
> 
> > > 
> 
> > > 
> 
> > > I met the same question! But I use ros2 ,what should I do ？
> 
> > 
> 
> > hi did u solved the problem yet? if so how? i also use ros2
> 
> 
> 
> I am not sure about ROS2, was using ROS at the time

Alright and did you successfully solve this problem? if so did you solve it using andelby mentioned method or others please tell me thanks.

Taoist-master, 2024-02-29:
> Alright and did you successfully solve this problem? if so did you solve it using andelby mentioned method or others please tell me thanks.

l use ros2 ,you should rename libDBow2.so or delete /usr/lib/libDBow2.so , only use /orb_slam2/Thirdparty/DBoW2/lib/libDBoW2.so