appliedAI-Initiative/orb_slam_2_ros #44: Not subscribing to Camera

MrMinemeet, 2019-10-08:
Hi,

I got the issue that the ORB-SLAM that I have in my workspace doesn't want to subscribe to my _/camera/image_raw_.
I have tried a few things where I used a video from a file, from a bag and from my webcam.
Still didn't get it to subscribe.

It built out of the box and also launches.

Here is a screenshot of the rqt node overview
[image]

lennarthaller, 2019-10-08:
Can you verify that your cam node is publishing the topic your orb_slam is subscribing to?

MrMinemeet, 2019-10-08:
@lennarthaller according to the documentation the ORB-SLAM2 should subscribe to ` /camera/image_raw` which the camera is publishing.
When using RQt and selecting ` /camera/image_raw` I  can see the video stream without a problem

lennarthaller, 2019-10-08:
I dont know how your launch file looks. What does rostopic list say?
I cannot imagine a reason why the node would not subscribe other then missconfiguration.

MrMinemeet, 2019-10-08:
Ok just found out why.
The remap in the launch file was the problem.
I just commented this line out:
` <remap from="/camera/image_raw" to="/camera/color/image_raw" />`

closed 2019-10-08 by MrMinemeet

barovjt, 2022-10-17:
@MrMinemeet  I tried to comment, and it yields to a crash. 

`[orb_slam2_mono-1] process has died [pid 19197, exit code -11, cmd /home/barov/b_ws/devel/lib/orb_slam2_ros/orb_slam2_ros_mono __name:=orb_slam2_mono __log:=/home/barov/.ros/log/24494d26-4e4b-11ed-bdfa-18c04d905575/orb_slam2_mono-1.log].
log file: /home/barov/.ros/log/24494d26-4e4b-11ed-bdfa-18c04d905575/orb_slam2_mono-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done`

I tried to publish directly to '/camera/rgb/image_raw' and i've got the same crash error. 
¿Do you have some advice?

MrMinemeet, 2022-10-18:
I mainly worked with it over the summer holidays, and this was already a few years back. If I remember correctly, there was a way to see all listening and providing services. Maybe testing out some, which make sens, may work.