appliedAI-Initiative/orb_slam_2_ros #45: Cannot view keep point see key point on rviz.

imohiosen, 2019-10-22:
Hi,

I am running orb_slam_2_ros to generate a point cloud and build an octomap out of it. However, even though the orb_slam_2_ros debug image shows key points being identified, the corresponding point could do not seem to be generated which contradicts log messages.

SNIPPET OF LOG OUTPUT
`
Waiting 3 seconds after advertising topics... done.
Hit space to toggle paused, or 's' to step.
New map created with 615 points
2.382931   Duration: 0.680575 / 173.477680               5151.03 
`

Here are the details of my setup?
dataset - rgbd_dataset_freiburg2_large_with_loop.bag
camera parameters file - TUM2.yaml

RVIZ SCREENSHOT
[image]

Relevant files.
[relevantfiles.zip](https://github.com/appliedAI-Initiative/orb_slam_2_ros/files/3753978/relevantfiles.zip)

Ruthrash, 2020-02-01:
In your "camera_setting".yaml file there's a parameter called DepthMapFactor, if it is high, keep it as 1. The topic is publishing the points but the scale is too small to see it.

lennarthaller, 2020-04-02:
Going to close this due to inactivity. Feel free to reopen if needed.

closed 2020-04-02 by lennarthaller

Hezhexi2002, 2023-01-24:
> Going to close this due to inactivity. Feel free to reopen if needed.

Hi,sorry to bother you here,but I have a similar problem when I try to run the orb_slam_2_ros on tum dataset.I use `roslaunch orb_slam2_ros orb_slam2_tum2_rgbd.launch` to launch the node,However,I don't know how to configure the dataset in the `orb_slam2_tum2_rgbd.launch` and there is no output when I open the rviz in another terminal:
[image]
so I want to ask what should I do to run the orb_slam_2_ros on the tum dataset,I would be very appreciated if you could give me some advice:-)