appliedAI-Initiative/orb_slam_2_ros #49: Maps cannot be saved multiple times

CodesHub, 2019-11-21:
hi, 
when I first run `rosservice call /orb_slam2_rgbd/save_map "name: 'map.bin'" `, it successfully saved the map, but when I called second time it failed and return `Error changing the call stack size; Aborting`. it seems SetCallStackSize() failed, how can I  save map twice?
Thanks!

CodesHub, 2019-11-21:
I found the bug!
in file System.cc  L156: `System::SetCallStackSize()`
` if (rlimit.rlim_cur < kNewStackSize) `
change to
` if (rlimit.rlim_cur <= kNewStackSize) `
it work fine, for sometime the 'rlimit.rlim_cur == kNewStackSize', it not should be consider  as an erro.

lennarthaller, 2019-11-27:
Can you please submit a Pull Request?

CodesHub, 2019-11-28:
not at all, I create a new pull request.

lennarthaller, 2019-11-28:
Perfect, thanks

closed 2019-11-28 by lennarthaller

abhayraw1, 2020-06-05:
Can't seem to locate where the maps are saved at?
I'm not quite sure what's wrong, the response from the service call and the output from the node suggests that the map has been saved. However if I run the find command `find . -name "*map*"` it doesn't return any thing of importance! `find` was run from the current `catkin_workspace` and it's the same directory from where the service was called too.

Here's the output node:

```
saving map file: map.bin ... done
[ INFO] [1591391667.031126800]: Map was saved as map.bin
```
Response from the service call:
```
rosservice call /orb_slam2_stereo/save_map "name: 'map.bin'"
success: True
```
Although I was able to save the file (or at least access it) by passing an absolute path to in the name!!
I'm just curious if those files were created or not?

lennarthaller, 2020-06-07:
From my README:
> You can replace "map.bin" with any file name you want. The file will be saved at ROS_HOME which is by default ~/.ros