appliedAI-Initiative/orb_slam_2_ros #53: Camera topics subscribed to for intel realsense D435 RGBD

JadTawil-theonly, 2019-12-16:
Hello,

The given launch file orb_slam2_d435_rgbd launch files seems to subscribe to the topics:

/camera/color/image_rect_color
/camera/depth/image_rect_raw

However, the depth/image_rect_raw is not actually aligned with the color/image_rect_color image.
In the realsense node, if align_depth is set to True, additional topics are published, one of which is:

/camera/aligned_depth_to_color/image_raw, which seems to be the depth image aligned with the color image (im not sure if this is the raw color image, or the rectified color image).

Wouldn't  the aligned depth image topic be the correct one to subscribe to? Since it's aligned with the color image being used to form the RGBD image?

lennarthaller, 2019-12-16:
I just checked the rs_rgbd.launch file from the realsense-ros repo, I think you are right. Good catch! 
I'll verify on Wednesday with a real sense and then fix it in the orb_slam launch files. I am also not sure which color image is the right now though.
I saw you opened Issues on the realsense repo, lets hope they answer quick ;)

JadTawil-theonly, 2019-12-17:
totally,

it's really weird, they publish a raw image, and a rectified image, but the intrinsics published report zero for distortion coeffs, which implies that the raw image is already rectified, leaving us with the question, what is the difference between the two image streams? Hopefully they answer

zainmehdi, 2020-08-20:
any update here ?