appliedAI-Initiative/orb_slam_2_ros #57: Opencv2.4.3 not correctly linked in CMakeLists.txt

KKKKKyle, 2020-02-28:
I encountered this issue when trying to use "cmake build" to build this package. The error output looks like this:

```
~/catkin_ws/src/orb_slam_2_ros/orb_slam2/lib/liborb_slam2_ros.so: undefined reference to `cvmake[2]: *** [~/hcilab/catkin_ws/devel/.private/orb_slam2_ros/lib/orb_slam2_ros/orb_slam2_ros_rgbd] Error 1
::_OutputArray::_OutputArray(cv::Mat const&)'
collect2: error: ld returned 1 exit status
make[1]: *** [CMakeFiles/orb_slam2_ros_rgbd.dir/all] Error 2
make[2]: *** [~/catkin_ws/devel/.private/orb_slam2_ros/lib/orb_slam2_ros/orb_slam2_ros_stereo] Error 1
make[1]: *** [CMakeFiles/orb_slam2_ros_stereo.dir/all] Error 2
make: *** [all] Error 2
```

so I googled and found a similar issue with cv_bridge here: [cv_bridge opencv not linked correctly](https://answers.ros.org/question/286128/cv_bridge-tutorial-example-not-compiling-in-ros-kinetic-and-ubuntu-1604/)

I tried those solutions but failed. Then I found [this project](https://github.com/PaolaArdon/Salt-Pepper/tree/master/orb_slam2) did similar thing so I added the following lines to your CMakeLists.txt:

```
target_link_libraries(${PROJECT_NAME}
${OpenCV_LIBS}
${EIGEN3_LIBS}
${PROJECT_SOURCE_DIR}/orb_slam2/Thirdparty/DBoW2/lib/libDBoW2.so
${Pangolin_LIBRARIES}
${PROJECT_SOURCE_DIR}/orb_slam2/Thirdparty/g2o/lib/libg2o.so
${Boost_LIBRARIES}
)
```
Then it compiled successfully with some warnings: 
```
Warnings   << orb_slam2_ros:make ~/catkin_ws/logs/orb_slam2_ros/build.make.006.log
/usr/bin/ld: warning: libopencv_core.so.2.4, needed by ~/hcilab/catkin_ws/src/orb_slam_2_ros/orb_slam2/lib/liborb_slam2_ros.so, may conflict with libopencv_core.so.3.2
/usr/bin/ld: warning: libopencv_core.so.2.4, needed by ~/hcilab/catkin_ws/src/orb_slam_2_ros/orb_slam2/lib/liborb_slam2_ros.so, may conflict with libopencv_core.so.3.2
/usr/bin/ld: warning: libopencv_core.so.2.4, needed by ~/hcilab/catkin_ws/src/orb_slam_2_ros/orb_slam2/lib/liborb_slam2_ros.so, may conflict with libopencv_core.so.3.2
```

I noticed that those libraries are "set" in your CMakeLists.txt but I am not sure if that can do the similar job as "target_link_libraries" as I am new to CMake.

Should this be a pull request?

lennarthaller, 2020-02-28:
Why are you using cmake build instead of catkin build?

KKKKKyle, 2020-03-01:
@lennarthaller Oops, that was a typo. I was using "catkin build"

KKKKKyle, 2020-03-01:
My bad... I was totally lost on that day. I re-downloaded your repository and put it under catkin_ws/src and this error won't appear with catkin build. However, I still encounter an issue. 

```
~/catkin_ws/src/ros_orb_slam2/orb_slam2/lib/liborb_slam2_ros.so: undefined reference to `cv::line(cv::Mat&, cv::Point_<int>, cv::Point_<int>, cv::Scalar_<double> const&, int, int, int)'
~/catkin_ws/src/ros_orb_slam2/orb_slam2/lib/liborb_slam2_ros.so: undefined reference to `cv::getTextSize(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int, double, int, int*)'
~catkin_ws/src/ros_orb_slam2/orb_slam2/lib/liborb_slam2_ros.so: undefined reference to `cv::_OutputArray::_OutputArray(cv::Mat const&)'
collect2: error: ld returned 1 exit status
make[2]: *** [~/catkin_ws/devel/.private/orb_slam2_ros/lib/orb_slam2_ros/orb_slam2_ros_stereo] Error 1
make[1]: *** [CMakeFiles/orb_slam2_ros_stereo.dir/all] Error 2
make: *** [all] Error 2
```
Guess I will open another issue for this.

closed 2020-03-01 by KKKKKyle