appliedAI-Initiative/orb_slam_2_ros #58: with OpenCV 2.4.3, fail to catkin build

KKKKKyle, 2020-03-01:
Thanks for any help in advance. 
 
I downloaded the repository and put it under catkin_ws/src. However, I encountered an issue with catkin build.

```
........
.......
.......
~/catkin_ws/src/ros_orb_slam2/orb_slam2/lib/liborb_slam2_ros.so: undefined reference to `cv::line(cv::Mat&, cv::Point_<int>, cv::Point_<int>, cv::Scalar_<double> const&, int, int, int)'
~/catkin_ws/src/ros_orb_slam2/orb_slam2/lib/liborb_slam2_ros.so: undefined reference to `cv::getTextSize(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int, double, int, int*)'
~/catkin_ws/src/ros_orb_slam2/orb_slam2/lib/liborb_slam2_ros.so: undefined reference to `cv::_OutputArray::_OutputArray(cv::Mat const&)'
collect2: error: ld returned 1 exit status
make[2]: *** [~/catkin_ws/devel/.private/orb_slam2_ros/lib/orb_slam2_ros/orb_slam2_ros_stereo] Error 1
make[1]: *** [CMakeFiles/orb_slam2_ros_stereo.dir/all] Error 2
make: *** [all] Error 2
```
It seems to be an issue with conflicting versions of opencv. The version installed on my end is 2.4.3.

Any suggestions? Thanks in advance!

darshb34, 2020-03-13:
@KKKKKyle can you share the full output log of the error?

KKKKKyle, 2020-03-26:
sry that I ran into this issue on another laptop which was not accessible due to Covid19... Will reopen this if I encounter this issue again in the future. Thx for helping out. Stay Healthy...

closed 2020-03-26 by KKKKKyle