appliedAI-Initiative/orb_slam_2_ros #65: Cannot load calibration from camera_info

tim-fan, 2020-04-13:
Using [orb_slam2_d435_rgbd.launch](https://github.com/appliedAI-Initiative/orb_slam_2_ros/blob/master/ros/launch/orb_slam2_d435_rgbd.launch), with param `load_calibration_from_cam` set `true`, system fails to launch (hangs on startup).

From what I can tell, the node hangs in [Node.cc#L239](https://github.com/appliedAI-Initiative/orb_slam_2_ros/blob/master/ros/src/Node.cc#L239), while waiting for a camera info message.

Looking further, it seems that `camera_info_topic_` is an empty string, which is probably why the camera info lookup never completes.

Note that `camera_info_topic_` is set in [RGBDNode.cc#L30](https://github.com/appliedAI-Initiative/orb_slam_2_ros/blob/master/ros/src/RGBDNode.cc#L30), in the RGBD node constructor, while the camera_info lookup occurs in the parent class [Node constructor](https://github.com/appliedAI-Initiative/orb_slam_2_ros/blob/master/ros/src/Node.cc#L24). 

So it appears that the camera info lookup occurs in the Node constructor, *before* the RGBD node constructor is able to set the correct `camera_info_topic_`, and so the lookup fails (hangs). 

I've worked around this by setting `camera_info_topic_` to `camera_info` in the Node constructor, just before the call to [LoadOrbParameters](https://github.com/appliedAI-Initiative/orb_slam_2_ros/blob/master/ros/src/Node.cc#L24). Then in `orb_slam2_d435_rgbd.launch` I remap from `camera_info` to the actual topic I am using. 

If you think this is an acceptable change, I can prepare a pull request. I feel this is preferable to the currently hard-coded camera info topics in the child class constructors, but maybe there's a better way to handle this.

lennarthaller, 2020-04-13:
Hey!
Yeah, I agree that the current implementation is really not beautiful. The problem is that all three SLAM types (mono, stereo and RGBD) use different camera topic names. Thats why the cam_info topic name is defined in the RGBD node and the look up happens in the node class since ints the same for all three.

A better solution might be to just subscribe to camera/camera_info from the node class and then remap the topic to the correct one in each launch file.

lennarthaller, 2020-04-22:
Should be fixed with the merge of #66

closed 2020-04-22 by lennarthaller