appliedAI-Initiative/orb_slam_2_ros #68: Expected DBoW Build/Linking Issue

CrYo-Dyne, 2020-04-30:
Hi @lennarthaller 

I was hoping you could perhaps help me with an issue I am having, similar to:

[https://github.com/appliedAI-Initiative/orb_slam_2_ros/issues/43](url)

I have modified the CMakeLists.txt according to the answer provided in the above issue and no longer get the build issues, however once I launch one of the example launch files I get the following error message:

> ROS_MASTER_URI=http://localhost:11311

process[orb_slam2_stereo-1]: started with pid [11258]
/home/rick/orb_slam_2_ros_ws/devel/lib/orb_slam2_ros/orb_slam2_ros_stereo: symbol lookup error: /home/rick/orb_slam_2_ros_ws/src/orb_slam_2_ros/orb_slam2/lib/liborb_slam2_ros.so: undefined symbol: _ZN5DBoW24FORB1LE
[orb_slam2_stereo-1] process has died [pid 11258, exit code 127, cmd /home/rick/orb_slam_2_ros_ws/devel/lib/orb_slam2_ros/orb_slam2_ros_stereo image_left/image_color_rect:=/mynteye/left_rect/image_rect image_right/image_color_rect:=/mynteye/right_rect/image_rect __name:=orb_slam2_stereo __log:=/home/rick/.ros/log/c414208e-8aac-11ea-84d0-c8f7506ba7d6/orb_slam2_stereo-1.log].
log file: /home/rick/.ros/log/c414208e-8aac-11ea-84d0-c8f7506ba7d6/orb_slam2_stereo-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done

So to me it seems as though there is a linking issue to the DBoW library as soon as it is launched. What's even more interesting is that the exact same error is obtained when running a rosrun ORB_SLAM2 Stereo example from the normal ORB_SLAM2 package.

I was hoping you might have some insight as to what could be the problem.

Thanks!

CrYo-Dyne, 2020-04-30:
Hi @lennarthaller 

It seems as though I might have come across the issue, I had a further look with `ldd` on the various libs produced during `catkin build`. Hopefully this could help with someone running into the same issue in the future.

First check to what orb_slam2_ros_stereo links with:
`~/orb_slam_2_ros_ws/devel/lib/orb_slam2_ros$ ldd orb_slam2_ros_stereo
	linux-vdso.so.1 (0x00007fffbd19e000)
	liborb_slam2_ros.so => /home/rick/orb_slam_2_ros_ws/src/orb_slam_2_ros/orb_slam2/lib/liborb_slam2_ros.so (0x00007f4e532af000)
	libcv_bridge.so => /opt/ros/melodic/lib/libcv_bridge.so (0x00007f4e53085000)
	libopencv_core.so.3.2 => /usr/lib/x86_64-linux-gnu/libopencv_core.so.3.2 (0x00007f4e52c4a000)
	libimage_transport.so => /opt/ros/melodic/lib/libimage_transport.so (0x00007f4e529cc000)
	libtf.so => /opt/ros/melodic/lib/libtf.so (0x00007f4e527a0000)
	libmessage_filters.so => /opt/ros/melodic/lib/libmessage_filters.so (0x00007f4e5259b000)
	libroscpp.so => /opt/ros/melodic/lib/libroscpp.so (0x00007f4e521fe000)
	librosconsole.so => /opt/ros/melodic/lib/librosconsole.so (0x00007f4e51fc8000)
	libdynamic_reconfigure_config_init_mutex.so => /opt/ros/melodic/lib/libdynamic_reconfigure_config_init_mutex.so (0x00007f4e51dc0000)
	libroscpp_serialization.so => /opt/ros/melodic/lib/libroscpp_serialization.so (0x00007f4e51bbd000)
	librostime.so => /opt/ros/melodic/lib/librostime.so (0x00007f4e5199c000)
	libboost_system.so.1.65.1 => /usr/lib/x86_64-linux-gnu/libboost_system.so.1.65.1 (0x00007f4e51797000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f4e51578000)
	libopencv_imgcodecs.so.3.4 => /usr/local/lib/libopencv_imgcodecs.so.3.4 (0x00007f4e512c4000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f4e50f3b000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f4e50b9d000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f4e50985000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f4e50594000)
	libDBoW2.so => /home/rick/slam_core_proto_ws/devel/lib/libDBoW2.so (0x00007f4e50387000)
	libg2o.so => /home/rick/orb_slam_2_ros_ws/src/orb_slam_2_ros/orb_slam2/Thirdparty/g2o/lib/libg2o.so (0x00007f4e500f6000)`

It can be seen from the above that libDBoW2.so is expected to be in the devel/lib location (not sure if this is correct?), whereas the g2o lib is what seems to me in the correct location of the src/../../Thirdparty location. Manually copying over the libDBoW2.so from the src/../../Thirdparty location to the catkin_ws's devel/lib directory seems to have solved the issue for now.

Thanks!

closed 2020-04-30 by CrYo-Dyne

lennarthaller, 2020-04-30:
Hey thanks for reporting this, this is helpful. I could't reproduce this on my machine which is strange. Same in the docker container. Are und on Ubuntu?

CrYo-Dyne, 2020-05-01:
No problem at all I hope it might help someone in a similar position :smile: 

Yes for me the setup is as follows:
Ubuntu 18.04
OpenCV 3.4.10/4.1.0

I had a previously installed DBoW2 and g2o before (manually building from source and installing) as well but I did clear them just to check if they could be causing the issue somehow.