appliedAI-Initiative/orb_slam_2_ros #69: Pose moving downwards ?

ChristofGroschke, 2020-05-02:
Hey there,

I'm a beginner using orbslam and ros. I have a question regarding the pose which is published to tf. 

I'm using a kinect camera on top of a turtlebot3. When moving forward on an even ground the position is moving downwards. Is there an easy way to keep the position on track ? Or is my configuration incorrect ?

[image]

https://pastebin.com/R3aZTR27

I'm grateful for every tip i can get :)

Ador2, 2020-10-06:
checkout `tf_orb_to_ros` in `Node.cc`
that rotates the orb frame to ROS frame, change it to match your system orientation

lennarthaller, 2021-01-16:
Could you fix your issue? Is your TF tree correct?

stephennacion06, 2021-05-18:
I'm having the same issue with rapsberry pi camera for monocular slam

stephennacion06, 2021-05-18:
[image]