appliedAI-Initiative/orb_slam_2_ros #70: Visual-Inertial ORB-SLAM

Sobolnikov, 2020-05-15:
Hello!

Thanks for very useful package!
I wonder if you plan to implement ros version of imu fusion into ORB us in https://github.com/YoujieXia/VI_ORB_SLAM2 or https://arxiv.org/pdf/1610.05949v1.pdf?

lennarthaller, 2020-05-15:
Hey! Its definitely on my feature list, but I am not sure if I am going to have the time to do it in the near feature. Are you interested in working on this?

Sobolnikov, 2020-05-15:
I am interested, but i think it will takes a lot of time to dive in. Maybe you already thought about it... Did you look at this packages:
https://github.com/YoujieXia/VI_ORB_SLAM2
https://github.com/ZuoJiaxing/Learn-ORB-VIO-Stereo-Mono
https://github.com/jingpang/LearnVIORB
Any comments regarding project structure and other implementation issues would help. As i understand all the packages based on this article https://arxiv.org/pdf/1610.05949v1.pdf.
Farther more i need few other features, maybe you show me direction for search. I interested in UGV, so it would possible to add some constraints on camera frame moving (in z direction in local frame for example). It would be good to use odometry information as well. There is possibility to use EKF (IMU+odom) as input, but i think there is a more "natural" and efficient way to fuse this data.
Did you try to use ORB with UGVs?
By the way, you may add RealSense D415 sensor in list (or in description) with the same config parameters as D435 - it works.

Sobolnikov, 2020-05-21:
I have started to learn ORB_SLAM IMU interaction. At first, i added new topic to ORB (PoseWithCovarianceStamped) in order to fuse pose from ORB with IMU orientation in robot_localization(ekf). It works, but it requires to tune pose covariance matrix.
Did you estimate covariance matrix of pose from ORB or maybe disperse of some components?