appliedAI-Initiative/orb_slam_2_ros #76: fatal error: orb_slam2_ros/SaveMap.h Not such File or Directory

juniorschen, 2020-07-11:
[image]
When i use catkin_make i recive this error, in orb_slam2_ros/ not have the SaveMap.h

juniorschen, 2020-07-12:
Somehow I managed to finish the build, however when trying to execute:
roslaunch orb_slam2_ros orb_slam2_r200_mono.launch
I have the following problem:
[image]
I'm following this short tutorial (I don't understand why we don't have a better step by step of these ROS packages, people "not very smart" like me end up having a lot of problems before using the tool minimum because of that):
https://www.hackadda.com/post/2020/5/17/using-orb-slam-ros-to-map-the-world

juniorschen, 2020-07-12:
Running turtlebot3 i have the image topics necessary:
[image]

juniorschen, 2020-07-12:
Complete Log:

[roslaunch][INFO] 2020-07-11 21:52:09,353: Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
[roslaunch][INFO] 2020-07-11 21:52:09,357: Done checking log file disk usage. Usage is <1GB.
[roslaunch][INFO] 2020-07-11 21:52:09,357: roslaunch starting with args ['/opt/ros/kinetic/bin/roslaunch', 'orb_slam2_ros', 'orb_slam2_r200_mono.launch']
[roslaunch][INFO] 2020-07-11 21:52:09,357: roslaunch env is {'TURTLEBOT_TYPE': 'turtlebot', 'ROS_DISTRO': 'kinetic', 'TURTLEBOT_STAGE_WORLD_FILE': '/opt/ros/kinetic/share/turtlebot_stage/maps/stage/maze.world', 'LESSOPEN': '| /usr/bin/lesspipe %s', 'TURTLEBOT_MAP_FILE': '/opt/ros/kinetic/share/turtlebot_navigation/maps/willow-2010-02-18-0.10.yaml', 'ROSLISP_PACKAGE_DIRECTORIES': '/root/catkin_ws/devel/share/common-lisp', 'LOGNAME': 'root', 'USER': 'root', 'HOME': '/root', 'PATH': '/opt/ros/kinetic/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games', 'CMAKE_PREFIX_PATH': '/root/catkin_ws/devel:/opt/ros/kinetic', 'LD_LIBRARY_PATH': '/root/catkin_ws/devel/lib:/opt/ros/kinetic/lib:/opt/ros/kinetic/lib/x86_64-linux-gnu', 'TURTLEBOT_STDR_MAP_FILE': '/opt/ros/kinetic/share/turtlebot_stdr/maps/sparse_obstacles.yaml', 'LANG': 'pt_BR.UTF-8', 'TERM': 'xterm-256color', 'SHELL': '/bin/bash', 'ROS_LOG_FILENAME': '/root/.ros/log/7606b5ea-c3ce-11ea-a879-080027e5d2a9/roslaunch-rosu-VirtualBox-6719.log', 'XAUTHORITY': '/home/rosu/.Xauthority', 'LANGUAGE': 'pt_BR:pt:en', 'TURTLEBOT_STACKS': 'hexagons', 'SHLVL': '1', 'TURTLEBOT_SIMULATION': 'false', 'ROS_ETC_DIR': '/opt/ros/kinetic/etc/ros', 'DISPLAY': ':0', 'TURTLEBOT_RAPP_PACKAGE_WHITELIST': '[rocon_apps, turtlebot_rapps]', 'TURTLEBOT_BATTERY': '/sys/class/power_supply/BAT1', 'SUDO_USER': 'rosu', 'TURTLEBOT_INTERACTIONS_LIST': '[turtlebot_bringup/admin.interactions, turtlebot_bringup/documentation.interactions, turtlebot_bringup/pairing.interactions, turtlebot_bringup/visualisation.interactions]', 'ROS_PYTHON_VERSION': '2', 'USERNAME': 'root', 'TURTLEBOT_RAPP_PACKAGE_BLACKLIST': '[]', 'PYTHONPATH': '/root/catkin_ws/devel/lib/python2.7/dist-packages:/opt/ros/kinetic/lib/python2.7/dist-packages', 'SUDO_UID': '1000', 'ROS_ROOT': '/opt/ros/kinetic/share/ros', 'TURTLEBOT_GAZEBO_MAP_FILE': '/opt/ros/kinetic/share/turtlebot_gazebo/maps/playground.yaml', 'PKG_CONFIG_PATH': '/root/catkin_ws/devel/lib/pkgconfig:/opt/ros/kinetic/lib/pkgconfig:/opt/ros/kinetic/lib/x86_64-linux-gnu/pkgconfig', 'ROS_PACKAGE_PATH': '/root/catkin_ws/src:/opt/ros/kinetic/share', 'TURTLEBOT_STAGE_MAP_FILE': '/opt/ros/kinetic/share/turtlebot_stage/maps/maze.yaml', 'TURTLEBOT_SERIAL_PORT': '/dev/kobuki', 'TURTLEBOT_BASE': 'kobuki', '_': '/opt/ros/kinetic/bin/roslaunch', 'SUDO_COMMAND': '/bin/su', 'SUDO_GID': '1000', 'TURTLEBOT_NAME': 'turtlebot', 'LESSCLOSE': '/usr/bin/lesspipe %s %s', 'OLDPWD': '/root/catkin_ws/src/orb_slam_2_ros/orb_slam2', 'ROS_IP': '192.168.15.10', 'PWD': '/root/catkin_ws/src/orb_slam_2_ros', 'TURTLEBOT_GAZEBO_WORLD_FILE': '/opt/ros/kinetic/share/turtlebot_gazebo/worlds/playground.world', 'ROS_MASTER_URI': 'http://192.168.15.10:11311', 'TURTLEBOT_3D_SENSOR': 'astra', 'ROS_VERSION': '1', 'MAIL': '/var/mail/root', 'LS_COLORS': 'rs=0:di=01;34:ln=01;36:mh=00:pi=40;33:so=01;35:do=01;35:bd=40;33;01:cd=40;33;01:or=40;31;01:mi=00:su=37;41:sg=30;43:ca=30;41:tw=30;42:ow=34;42:st=37;44:ex=01;32:*.tar=01;31:*.tgz=01;31:*.arc=01;31:*.arj=01;31:*.taz=01;31:*.lha=01;31:*.lz4=01;31:*.lzh=01;31:*.lzma=01;31:*.tlz=01;31:*.txz=01;31:*.tzo=01;31:*.t7z=01;31:*.zip=01;31:*.z=01;31:*.Z=01;31:*.dz=01;31:*.gz=01;31:*.lrz=01;31:*.lz=01;31:*.lzo=01;31:*.xz=01;31:*.bz2=01;31:*.bz=01;31:*.tbz=01;31:*.tbz2=01;31:*.tz=01;31:*.deb=01;31:*.rpm=01;31:*.jar=01;31:*.war=01;31:*.ear=01;31:*.sar=01;31:*.rar=01;31:*.alz=01;31:*.ace=01;31:*.zoo=01;31:*.cpio=01;31:*.7z=01;31:*.rz=01;31:*.cab=01;31:*.jpg=01;35:*.jpeg=01;35:*.gif=01;35:*.bmp=01;35:*.pbm=01;35:*.pgm=01;35:*.ppm=01;35:*.tga=01;35:*.xbm=01;35:*.xpm=01;35:*.tif=01;35:*.tiff=01;35:*.png=01;35:*.svg=01;35:*.svgz=01;35:*.mng=01;35:*.pcx=01;35:*.mov=01;35:*.mpg=01;35:*.mpeg=01;35:*.m2v=01;35:*.mkv=01;35:*.webm=01;35:*.ogm=01;35:*.mp4=01;35:*.m4v=01;35:*.mp4v=01;35:*.vob=01;35:*.qt=01;35:*.nuv=01;35:*.wmv=01;35:*.asf=01;35:*.rm=01;35:*.rmvb=01;35:*.flc=01;35:*.avi=01;35:*.fli=01;35:*.flv=01;35:*.gl=01;35:*.dl=01;35:*.xcf=01;35:*.xwd=01;35:*.yuv=01;35:*.cgm=01;35:*.emf=01;35:*.ogv=01;35:*.ogx=01;35:*.aac=00;36:*.au=00;36:*.flac=00;36:*.m4a=00;36:*.mid=00;36:*.midi=00;36:*.mka=00;36:*.mp3=00;36:*.mpc=00;36:*.ogg=00;36:*.ra=00;36:*.wav=00;36:*.oga=00;36:*.opus=00;36:*.spx=00;36:*.xspf=00;36:'}
[roslaunch][INFO] 2020-07-11 21:52:09,357: starting in server mode
[roslaunch.parent][INFO] 2020-07-11 21:52:09,358: starting roslaunch parent run
[roslaunch][INFO] 2020-07-11 21:52:09,358: loading roscore config file /opt/ros/kinetic/etc/ros/roscore.xml
[roslaunch][INFO] 2020-07-11 21:52:09,550: Added core node of type [rosout/rosout] in namespace [/]
[roslaunch.config][INFO] 2020-07-11 21:52:09,550: loading config file /root/catkin_ws/src/orb_slam_2_ros/ros/launch/orb_slam2_r200_mono.launch
[roslaunch][INFO] 2020-07-11 21:52:09,591: Added node of type [orb_slam2_ros/orb_slam2_ros_mono] in namespace [/]
[roslaunch][INFO] 2020-07-11 21:52:09,591: ... selected machine [] for node of type [orb_slam2_ros/orb_slam2_ros_mono]
[roslaunch.pmon][INFO] 2020-07-11 21:52:09,592: start_process_monitor: creating ProcessMonitor
[... cut ...]
[roslaunch.pmon][INFO] 2020-07-11 21:52:09,795: ProcessMonitor._post_run <ProcessMonitor(ProcessMonitor-1, started daemon 139970050221824)>: remaining procs are [<roslaunch.nodeprocess.LocalProcess object at 0x7f4d516abad0>]
[roslaunch.pmon][INFO] 2020-07-11 21:52:09,796: ProcessMonitor exit: killing orb_slam2_mono-1
[roslaunch][INFO] 2020-07-11 21:52:09,796: [orb_slam2_mono-1] killing on exit
[roslaunch.pmon][INFO] 2020-07-11 21:52:09,801: ProcessMonitor exit: cleaning up data structures and signals
[roslaunch.pmon][INFO] 2020-07-11 21:52:09,801: ProcessMonitor exit: pmon has shutdown
[rospy.core][INFO] 2020-07-11 21:52:09,813: signal_shutdown [atexit]

juniorschen, 2020-07-12:
Well after a torturous day of trying (I even tried to configure it by the main ORB_SLAM project but the hell of my virtual machine always caused memory problems when building), I finally got to the point of the tutorial I used, and how did I do it? I simply deleted the clone from the repository and did it again and everything works, sometimes I think that hell does not exist, until I have to work with problems like this in linux / ros

Wingmore, 2020-08-27:
> When i use catkin_make i recive this error, in orb_slam2_ros/ not have the SaveMap.h

How did you fix this issue? I'm getting the same one

juniorschen, 2020-08-27:
@Wingmore 
I deleted the package, went to the real project and installed all the dependencies he mentions and performed the build, after that I downloaded the ros package again and it worked

git: https://github.com/raulmur/ORB_SLAM2
dependencies:
[image]

closed 2020-09-24 by juniorschen

mcmchammer, 2021-09-08:
Hi - I'm hitting the same issue, but missing a root cause in this thread ... any insights regarding what exactly is causing this error?

AnaRhisT94, 2022-03-25:
> In case anyone is running into this, this error arises from missing message generation dependencies for the build targets in CMakeLists.txt; add `${PROJECT_NAME}_generate_messages_cpp` to the dependencies as follows
> 
> ```
> ...
> add_dependencies (${PROJECT_NAME}_mono ${PROJECT_NAME} ${PROJECT_NAME}_gencfg ${PROJECT_NAME}_generate_messages_cpp)
> ...
> add_dependencies (${PROJECT_NAME}_stereo ${PROJECT_NAME} ${PROJECT_NAME}_gencfg ${PROJECT_NAME}_generate_messages_cpp)
> ...
> add_dependencies (${PROJECT_NAME}_rgbd ${PROJECT_NAME} ${PROJECT_NAME}_gencfg ${PROJECT_NAME}_generate_messages_cpp)
> ...
> ```

This actually solved my issue and I successfully built catkin.
Here the complete CMAKELists.txt file: (using ROS Melodic)

```
cmake_minimum_required(VERSION 2.8.3)
project(orb_slam2_ros)

IF(NOT CMAKE_BUILD_TYPE)
  SET(CMAKE_BUILD_TYPE Release)
ENDIF()

# Check for c++11 support
INCLUDE(CheckCXXCompilerFlag)
CHECK_CXX_COMPILER_FLAG("-std=c++11" COMPILER_SUPPORTS_CXX11)
CHECK_CXX_COMPILER_FLAG("-std=c++0x" COMPILER_SUPPORTS_CXX0X)
IF(COMPILER_SUPPORTS_CXX11)
  set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++11")
   add_definitions(-DCOMPILEDWITHC11)
   message(STATUS "Using flag -std=c++11.")
elseif(COMPILER_SUPPORTS_CXX0X)
   set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++0x")
   add_definitions(-DCOMPILEDWITHC0X)
   message(STATUS "Using flag -std=c++0x.")
ELSE()
  MESSAGE(ERROR "The compiler ${CMAKE_CXX_COMPILER} has no C++11 support. Please use a different C++ compiler.")
ENDIF()

LIST(APPEND CMAKE_MODULE_PATH ${PROJECT_SOURCE_DIR}/orb_slam2/cmake_modules)

find_package (catkin REQUIRED COMPONENTS
roscpp
rospy
std_msgs
cv_bridge
image_transport
tf
sensor_msgs
dynamic_reconfigure
message_generation
)

find_package(OpenCV 3.0 QUIET)
if(NOT OpenCV_FOUND)
   find_package(OpenCV 2.4.3 QUIET)
   if(NOT OpenCV_FOUND)
      message(FATAL_ERROR "OpenCV > 2.4.3 not found.")
   endif()
endif()

find_package(Eigen3 3.1.0 REQUIRED)
find_package(Pangolin REQUIRED)

set (DYNAMIC_RECONFIGURE_PATH ros/config/dynamic_reconfigure.cfg)
execute_process(COMMAND chmod a+x ${DYNAMIC_RECONFIGURE_PATH}
        WORKING_DIRECTORY ${PROJECT_SOURCE_DIR}
        RESULT_VARIABLE cmd_result
        OUTPUT_VARIABLE cmd_ver)
message(STATUS "Chmod a+x the dynamic_reconfigure file")
[... 122 lines cut ...]
  RUNTIME DESTINATION ${CATKIN_PACKAGE_BIN_DESTINATION}
)

install(TARGETS ${PROJECT_NAME} DBoW2 g2o
  ARCHIVE DESTINATION ${CATKIN_PACKAGE_LIB_DESTINATION}
  LIBRARY DESTINATION ${CATKIN_PACKAGE_LIB_DESTINATION}
  RUNTIME DESTINATION ${CATKIN_PACKAGE_BIN_DESTINATION}
)

install(DIRECTORY ros/launch/
  DESTINATION ${CATKIN_PACKAGE_SHARE_DESTINATION}/ros/launch
)

install(DIRECTORY ros/config/
        DESTINATION ${CATKIN_PACKAGE_SHARE_DESTINATION}/ros/config
)

install(DIRECTORY orb_slam2/config/
  DESTINATION ${CATKIN_PACKAGE_SHARE_DESTINATION}/orb_slam2/config
)

install(DIRECTORY orb_slam2/Vocabulary/
  DESTINATION ${CATKIN_PACKAGE_SHARE_DESTINATION}/orb_slam2/Vocabulary
)
```

arjunskumar, 2024-03-07:
@AnaRhisT94 Thank you _/\_