appliedAI-Initiative/orb_slam_2_ros #77: Unable to find package "orb_slam2_ros "  in pkg list when using Eloquent branch

AvinashRamashray, 2020-07-15:
Hi,

I am new to ROS world, I want to perform orb-slam2 with ROS2 .
So I created a colcon work space and cloned this repository and checkout to eloquent branch 
Steps performed:
1.Build this repository with colcon build and it was successful.
2.I tried to launch orb_slam2_d435_mono_launch.py directly but end up getting this error
    
avinash@avinash-G3-3590:~/eloquent_ws/src/orb_slam_2_ros/ros/launch$ ros2 launch orb_slam2_d435_mono_launch.py 
[INFO] [launch]: All log files can be found below /home/avinash/.ros/log/2020-07-15-10-27-01-428873-avinash-G3-3590-13890
[INFO] [launch]: Default logging verbosity is set to INFO
[ERROR] [launch]: Caught exception in launch (see debug for traceback): Caught exception when trying to load file of format [py]: "package 'orb_slam2_ros' not found, searching: ['/opt/ros/eloquent']"
avinash@avinash-G3-3590:~/eloquent_ws/src/orb_slam_2_ros/ros/launch$ 

Why this package  'orb_slam2_ros' not created?
Please guide what I am doing wrong or missing something.
If possible please update readme for how to use this repo  with ROS2 to achieve vSLAM
Thanks in advance.

AvinashRamashray, 2020-07-15:
Got it.

closed 2020-07-15 by AvinashRamashray

peakyquest, 2024-11-20:
What ros2 version you are using what other dependencies you have installed