appliedAI-Initiative/orb_slam_2_ros #83: Map Point Vector Empty with Stereo Setup

segozcan, 2020-09-18:
Hello, 
I am working on orb_slam with a stereo camera setup. I launch the following launch file:

> <launch>
  <node name="orb_slam2_stereo" pkg="orb_slam2_ros"
      type="orb_slam2_ros_stereo" output="screen">

       <remap from="image_left/image_color_rect" to="/camera/left/image_rect" />
       <remap from="image_right/image_color_rect" to="/camera/right/image_rect" />
       <remap from="image_left/camera_info" to="/camera/left/camera_info" />
       <param name="publish_pointcloud" type="bool" value="true" />
       <param name="publish_pose" type="bool" value="true" />
       <param name="localize_only" type="bool" value="false" />
       <param name="reset_map" type="bool" value="false" />

       
       <param name="load_map" type="bool" value="false" />
       <param name="map_file" type="string" value="map.bin" />
       <param name="voc_file" type="string" value="$(find orb_slam2_ros)/orb_slam2/Vocabulary/ORBvoc.txt" />

       <param name="pointcloud_frame_id" type="string" value="map" />
       <param name="camera_frame_id" type="string" value="camera" />
       <param name="min_num_kf_in_map" type="int" value="5" />

       
       <param name="/ORBextractor/nFeatures" type="int" value="8000" />
       <param name="/ORBextractor/scaleFactor" type="double" value="1.2" />
       <param name="/ORBextractor/nLevels" type="int" value="8" />
       <param name="/ORBextractor/iniThFAST" type="int" value="20" />
       <param name="/ORBextractor/minThFAST" type="int" value="6" />

       
       
       <param name="camera_fps" type="int" value="30" />
       
       <param name="camera_rgb_encoding" type="bool" value="true" />
       
       <param name="ThDepth" type="double" value="40.0" />
       
       <param name="depth_map_factor" type="double" value="1.0" />

        
        
       <param name="load_calibration_from_cam" type="bool" value="true" />
       
      <param name="camera_fx" type="double" value="1708.789386" />
      <param name="camera_fy" type="double" value="1705.648225" />
      <param name="camera_cx" type="double" value="827.374992" />
      <param name="camera_cy" type="double" value="559.485285" />
       
      <param name="camera_k1" type="double" value="-0.164735" />
      <param name="camera_k2" type="double" value="0.098101" />
      <param name="camera_p1" type="double" value="0.001468" />
      <param name="camera_p2" type="double" value="0.004257" />
      
      <param name="camera_baseline" type="double" value="14.81676006" />
  </node>
</launch>

In rqt_graph, topics seem to be properly subscribed. I get the error saying that the map point vector is empty and the system resets afterwards:

> Track lost soon after initialisation, reseting...
Map point vector is empty!
System Reseting
Reseting Local Mapper... done
Reseting Loop Closing... done
Reseting Database... done
New map created with 0 points

I do not understand what the problem is. For monocular setups I've come across issues that recommend moving the camera to initialize mapping but that did not work. Any help would be appreciated.

lennarthaller, 2021-01-16:
The resetting and the message "New map created with 0 points" tell me that the algorithm is not able to find any features which leads to the message of the map point vector being empty.
How is your environment? Is it simulated? Are there plenty color gradients in the overlapping FoV of both cameras?

closed 2021-01-16 by lennarthaller

JulioPlaced, 2021-04-19:
I'm having the same issue with a simulated environment in Gazebo with plenty of color gradients and textures (AWS house), using stereo cameras.