appliedAI-Initiative/orb_slam_2_ros #90: /camera/image_raw not subscribing to /videofile/image_raw

stephennacion06, 2020-10-25:
Hello, I am trying to do **roslaunch orb_slam2_ros orb_slam2_r200_mono.launch /camera/image_raw:=/videofile/image_raw** but there is still no images displaying both rviz and image view to orb slam 2

stephennacion06, 2020-10-25:
[image]

lennarthaller, 2020-10-25:
Have you looked at the terminal log of the node? Seems like #2. Mono slam from a video file where the slam can't initialize is gonna be tough.

stephennacion06, 2020-10-30:
Okay thank you for the reply and for your work. The issue is fix by changing the /camera/image_raw to /videofile/image_raw inside the Mononode.cc
[image]

closed 2020-10-30 by stephennacion06