appliedAI-Initiative/orb_slam_2_ros #101: Use odometry in mapping mode

cesar-lopez-mar, 2021-01-27:
Hi,

I have tested the algorithm and the localization during mapping suffers from drift. I already checked the camera settings and I am using an RGB-D camera. A way to reduce the drift is to use external odometry data while creating the map. I know rtab map has that feature, is this something that is within the developing roadmap of this package?

lennarthaller, 2021-01-27:
> Localization during mapping suffers from drift

Loop closure should fix (most of) it. How many loops do you have and how large is the mapped area?

> A way to reduce the drift is to use external odometry data

How do you mean "external"? Like wheel odometry? This will not fix drift (as its in itself very prone to drifting) but whill make the local SLAM more robust. Right now using wheel odometry is not supported.

cesar-lopez-mar, 2021-02-11:
Sorry for my late reply, I had two loops but I used a rosbag to loop those several times. Area was a living room around 30m2. 
With external odometry I mean more another localization method, like AMCL providing an odometry message which I can use to tell orb slam where the robot  is (with high confidence) when making the map. Then, I am planning to use only ORB slam in localization mode.