appliedAI-Initiative/orb_slam_2_ros #102: I am unable to run this package using intel realsense D435.

MarzanShuvo, 2021-02-05:
I am very beginner.I have build this ros package. Firstly I tried to launch `roslaunch realsense2_camera rs_rgbd.launch` for the subscribe topics. After running this, I run `roslaunch orb_slam2_ros orb_slam2_d435_mono.launch`. At first, It was printing **Map point vector is empty!**. After moving the camera to and fro, it was printing **Could not find a connection between 'camera_link' and 'base_link' because they are not part of the same tree.Tf has two or more unconnected trees**. I spent almost two days for getting it right. Would you like to tell me what I am going wrong? What steps should I follow to get it right?

lennarthaller, 2021-03-07:
Your tf tree is disconnected (there is no transform from camera_link to base_link). For an introduction to the tf2 concepts check out [this link](http://wiki.ros.org/tf2/Tutorials).
For your case it seems like you need a static transform from cam to base_link with something linke the static transform publisher. Check out the [ros_tf2](http://wiki.ros.org/tf2_ros) (under the Tools section).

orb_slam2_ros will give you the transformation from the map frame to camera_link out of the box.

closed 2021-03-07 by lennarthaller

wrmichaelmak, 2021-03-24:
Sorry for reopening this issue. But what is the plan of action to solve the problem? Do we need to add a tf broadcaster?

clydemcqueen, 2021-03-24:
You can publish the transform using a tf broadcaster in C++ or Python code, or using the `robot_state_publisher` node with an URDF file. You can also use the `static_transform_publisher` utility to publish a single static transform:

`/opt/ros/foxy/lib/tf2_ros/static_transform_publisher 0 0 0 0 0 0 base_link camera_link`

There's lots more info on ROS Answers, e.g., this question: https://answers.ros.org/question/208002/static-transformation-publisher/

wrmichaelmak, 2021-03-24:
Cheers. That was what I thought as well! Thank you for the validation and quick response!

ozumipl, 2022-08-11:
> I am very beginner.I have build this ros package. Firstly I tried to launch `roslaunch realsense2_camera rs_rgbd.launch` for the subscribe topics. After running this, I run `roslaunch orb_slam2_ros [...]

Hi,
Currently I am having same problem and I did not understand how to solve problem. Here what I did:
I did run that: `roslaunch realsense2_camera rs_rgbd.launch `
After that I did run it: `roslaunch orb_slam2_ros orb_slam2_d435_mono.launch`

As an output log, I did obtain the following lines:
`... logging to /home/jetsonnano2/.ros/log/279dc6da-196e-11ed-a084-3413e863a5ce/roslaunch-jetsonnano2-desktop-24375.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://10.200.76.132:33919/

SUMMARY
========

PARAMETERS
 * /orb_slam2_mono/ORBextractor/iniThFAST: 20
 * /orb_slam2_mono/ORBextractor/minThFAST: 7
 * /orb_slam2_mono/ORBextractor/nFeatures: 2000
 * /orb_slam2_mono/ORBextractor/nLevels: 8
 * /orb_slam2_mono/ORBextractor/scaleFactor: 1.2
 * /orb_slam2_mono/camera_cx: 328.001068115
 * /orb_slam2_mono/camera_cy: 241.310317993
 * /orb_slam2_mono/camera_fps: 30
 * /orb_slam2_mono/camera_frame_id: camera_link
 * /orb_slam2_mono/camera_fx: 615.670715332
 * /orb_slam2_mono/camera_fy: 615.962158203
 * /orb_slam2_mono/camera_k1: 0.0
 * /orb_slam2_mono/camera_k2: 0.0
 * /orb_slam2_mono/camera_k3: 0.0
 * /orb_slam2_mono/camera_p1: 0.0
 * /orb_slam2_mono/camera_p2: 0.0
 * /orb_slam2_mono/camera_rgb_encoding: True
 * /orb_slam2_mono/load_calibration_from_cam: False
 * /orb_slam2_mono/load_map: False
 * /orb_slam2_mono/localize_only: False
 * /orb_slam2_mono/map_file: map.bin
 * /orb_slam2_mono/min_num_kf_in_map: 5
 * /orb_slam2_mono/pointcloud_frame_id: map
 * /orb_slam2_mono/publish_pointcloud: True
 * /orb_slam2_mono/publish_pose: True
 * /orb_slam2_mono/reset_map: False
 * /orb_slam2_mono/voc_file: /home/jetsonnano2...
 * /rosdistro: melodic
 * /rosversion: 1.14.13

NODES
  /
    orb_slam2_mono (orb_slam2_ros/orb_slam2_ros_mono)

ROS_MASTER_URI=http://10.200.76.132:11311/

process[orb_slam2_mono-1]: started with pid [24392]

ORB-SLAM2 Copyright (C) 2014-2016 Raul Mur-Artal, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

OpenCV version : 4.1.1
Major version : 4
Minor version : 1
Subminor version : 1
Input sensor was set to: Monocular

Loading ORB Vocabulary.
Vocabulary loaded!

Camera Parameters: 
- fx: 615.671
- fy: 615.962
- cx: 328.001
- cy: 241.31
- k1: 0
- k2: 0
- p1: 0
- p2: 0
- fps: 30
- bf: 1.77965e-43
- color order: RGB (ignored if grayscale)

ORB Extractor Parameters: 
- Number of Features: 2000
- Scale Levels: 8
- Scale Factor: 1.2
- Initial Fast Threshold: 20
- Minimum Fast Threshold: 7
Enable localization only: false
Map point vector is empty!
Map point vector is empty!
Map point vector is empty!
Map point vector is empty!
Map point vector is empty!
Map point vector is empty!
Map point vector is empty!

`
So, is there any specific solution to that? Furthermore, I did not get anything on rviz also. Almost all topics (generated after above runs) are empty such as pose etc.