appliedAI-Initiative/orb_slam_2_ros #104: I am unable to run using ZED2

Hyper-Devil, 2021-03-06:
I modified the parameters such as Camera topics and camera_fx in the launch file, and it appears at run time: 
New map created with 268 points
[ WARN] [1615002039.816134264]: "zed2_camera_center" passed to lookupTransform argument target_frame does not exist. 
[ WARN] [1615002039.817486992]: "zed2_camera_center" passed to lookupTransform argument target_frame does not exist. 
[ WARN] [1615002039.872931648]: "zed2_camera_center" passed to lookupTransform argument target_frame does not exist. 
[ WARN] [1615002039.872990663]: "zed2_camera_center" passed to lookupTransform argument target_frame does not exist. 
[ WARN] [1615002039.919124112]: "zed2_camera_center" passed to lookupTransform argument target_frame does not exist. 
[ WARN] [1615002039.919181140]: "zed2_camera_center" passed to lookupTransform argument target_frame does not exist. 
There is no UI interface, how can I solve it?

Thanks for your answer.

lennarthaller, 2021-03-07:
The launch file specifies the camera frame as:
`<param name="camera_frame_id" type="string" value="zed2_camera_center" />`
Make sure that this is matching your configuration.

closed 2021-03-07 by lennarthaller

Hyper-Devil, 2021-03-08:
@lennarthaller  Thank you for your answer. As I am still a beginner, I sincerely hope you can give me more help.

I think the "camera_frame_id" is correct. Maybe there is something wrong with "target_frame_id", but I don't know how to solve it.

The TF tree maintained by zed_wrapper:
[image]

The launch file I use:
```
<param name="pointcloud_frame_id" type="string" value="map" />
<param name="camera_frame_id" type="string" value="zed2_camera_center" />
<param name="target_frame_id" type="string" value="base_link" />
```
Launch file running result and new tf tree:
[image]
[image]

Could you give me some advice?