appliedAI-Initiative/orb_slam_2_ros #105: Can't get the tf date.

zhuangxiaopi, 2021-03-14:
The terminal shows : 
[ WARN] [1615737098.457917451]: "kinect2_rgb_optical_frame" passed to lookupTransform argument target_frame does not exist.

I use kinectv2 , The following is my modification in the launch file
`      <remap from="/camera/rgb/image_raw" to="/kinect2/hd/image_color_rect" />
       <remap from="/camera/depth_registered/image_raw" to="/kinect2/hd/image_depth_rect" />

       <param name="camera_frame_id" type="string" value="kinect2_rgb_optical_frame" />`
Thx.

lennarthaller, 2021-03-14:
Please verify that your TF tree is set up correctly.

closed 2021-03-14 by lennarthaller

zhuangxiaopi, 2021-03-15:
USE KinectV2 can use below command 
1.Pulish kinect tf
`roslaunch kinect2_bridge kinect2_bridge.launch publish_tf:=true`
2.create relation with
`rosrun tf static_transform_publisher 0 0 0 0 0 0 base_link kinect2_link 100 
`
3.Modify file orb_slam2_r200_rgbd.launch
4.
`roslaunch orb_slam2_ros orb_slam2_r200_rgbd.launch`