appliedAI-Initiative/orb_slam_2_ros #133: Visualize and save the trajectory

canglangzhige, 2022-04-28:
I can not visualize the trajectory in rviz by add topic. If there is not the topic?
Can I get the saved trajectory?

geturin, 2022-05-03:
you can use the  [hector_trajectory_server](http://wiki.ros.org/hector_trajectory_server) 

in your launch file, like:

<node pkg="hector_trajectory_server" type="hector_trajectory_server" name="trajectory_server_orb_slam2" output="screen" ns="orb_slam2_ros" >
        <param name="/target_frame_name" value="/map" />
        <param name="/source_frame_name" value="/base_link" />
        <param name="/trajectory_update_rate" value="20.0" />
        <param name="/trajectory_publish_rate" value="20.0" />
    </node>

canglangzhige, 2022-05-03:
OK. Thank you!