appliedAI-Initiative/orb_slam_2_ros #134: Tf has two or more unconnected trees

Garrison-ctrl, 2022-05-04:
`[ WARN] [1651662336.443545691]: Could not find a connection between 'camera_link' and 'base_link' because they are not part of the same tree.Tf has two or more unconnected trees.` so that the pointcloud has no result

Casey-J-Wolcott, 2022-05-13:
I'm having the same problem, I'm trying to add /map_points to rviz to get a visual map, but between world_emu and map.

Garrison-ctrl, 2022-05-21:
> 

I have solved this problem,but the way is a little strange,I use the `roslaunch realsense2_camera rs_rgbd.launch` to start up the d435i,and use the `roslaunch orb_slam2_ros orb_slam2_d435_mono.launch`,at first the terminal will show `the map is empty`,after a little second,you can open rviz and add the pointcloud and camera,you will find pointcloud.

rzhao1024, 2023-03-22:
Hi! I encountered the same problem. When I start up the d435i using roslaunch orb_slam2_ros orb_slam2_d435_rgbd.launch, the terminal also shows [ WARN] [1651662336.443545691]: Could not find a connection between 'camera_link' and 'base_link' because they are not part of the same tree.Tf has two or more unconnected trees. 
How to solve this problem?