appliedAI-Initiative/orb_slam_2_ros #139: Slam doesn’t detect movement

MagdaZal, 2022-11-02:
Hi,

I have a problem with the motion detection of the program. I have a drone with installed GoPro Hero 6 which drives along the coast. ORB SLAM reads the topic from camera correctly and creates a point cloud (I see preview in Rviz), but the point cloud is overwritten at one point. It’s like the SLAM didn’t read the drone’s motion. I can only see Pose rotating left-right, but he’s in one place.

The same situation occurs during tests in the room when I only move the Gopro camera. A point cloud forms in one place. If the camera is moved forward/rear, the ORB SLAM does not read this motion or the pose moves left-right.

All topics in ROS are made correctly.

I can’t find the cause of the problem. I’d be grateful for your help 😊

*************************************************************
Commands in terminal:

```
roscore
roslaunch video_stream_opencv webcamGOPRO.launch
roslaunch orb_slam2_ros GOPRO.launch
```
*************************************************************
Terminal window with SLAM:
```
magda@magda-VirtualBox:~$ roslaunch orb_slam2_ros GOPRO.launch
... logging to /home/magda/.ros/log/1fdb58d2-5a7c-11ed-8965-080027066cc7/roslaunch-magda-VirtualBox-21299.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://magda-VirtualBox:35645/

SUMMARY
========

PARAMETERS
 * /orb_slam2_mono/ORBextractor/iniThFAST: 20
 * /orb_slam2_mono/ORBextractor/minThFAST: 7
 * /orb_slam2_mono/ORBextractor/nFeatures: 2000
 * /orb_slam2_mono/ORBextractor/nLevels: 8
 * /orb_slam2_mono/ORBextractor/scaleFactor: 1.2
 * /orb_slam2_mono/camera_cx: 0.0
 * /orb_slam2_mono/camera_cy: 0.0
 * /orb_slam2_mono/camera_fps: 60
 * /orb_slam2_mono/camera_frame_id: camera_link
 * /orb_slam2_mono/camera_fx: 0.0
 * /orb_slam2_mono/camera_fy: 0.0
 * /orb_slam2_mono/camera_k1: 0.0
 * /orb_slam2_mono/camera_k2: 0.0
 * /orb_slam2_mono/camera_k3: 0.0
 * /orb_slam2_mono/camera_p1: 0.0
 * /orb_slam2_mono/camera_p2: 0.0
 * /orb_slam2_mono/camera_rgb_encoding: True
 * /orb_slam2_mono/load_calibration_from_cam: True
 * /orb_slam2_mono/load_map: False
 * /orb_slam2_mono/localize_only: False
 * /orb_slam2_mono/map_file: map.bin
 * /orb_slam2_mono/min_num_kf_in_map: 5
 * /orb_slam2_mono/pointcloud_frame_id: map
 * /orb_slam2_mono/publish_pointcloud: True
 * /orb_slam2_mono/publish_pose: True
 * /orb_slam2_mono/reset_map: False
 * /orb_slam2_mono/voc_file: /home/magda/catki...
 * /rosdistro: melodic
 * /rosversion: 1.14.13

NODES
  /
    orb_slam2_mono (orb_slam2_ros/orb_slam2_ros_mono)
    rviz (rviz/rviz)

ROS_MASTER_URI=http://localhost:11311

process[orb_slam2_mono-1]: started with pid [21314]
process[rviz-2]: started with pid [21315]
[ INFO] [1667373565.654140140]: Listening for camera info on topic /webcam/camera_info

ORB-SLAM2 Copyright (C) 2014-2016 Raul Mur-Artal, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

OpenCV version : 3.2.0
Major version : 3
Minor version : 2
Subminor version : 0
Input sensor was set to: Monocular

Loading ORB Vocabulary.
Vocabulary loaded!

Camera Parameters: 
- fx: 1230.04
- fy: 1225.98
- cx: 809.051
- cy: 560.795
- k1: -0.410783
- k2: 0.160178
- p1: -0.00019456
- p2: -0.000719679
- fps: 60
- bf: 0
- color order: RGB (ignored if grayscale)

ORB Extractor Parameters: 
- Number of Features: 2000
- Scale Levels: 8
- Scale Factor: 1.2
- Initial Fast Threshold: 20
- Minimum Fast Threshold: 7
Enable localization only: false
Map point vector is empty!
Map point vector is empty!
QObject::connect: Cannot queue arguments of type 'QVector<int>'
(Make sure 'QVector<int>' is registered using qRegisterMetaType().)
QObject::connect: Cannot queue arguments of type 'QVector<int>'
(Make sure 'QVector<int>' is registered using qRegisterMetaType().)
Map point vector is empty!
Map point vector is empty!
New Map created with 218 points
[ WARN] [1667373571.956543954]: "camera_link" passed to lookupTransform argument target_frame does not exist. 
[ WARN] [1667373571.957448246]: "camera_link" passed to lookupTransform argument target_frame does not exist. 
[ WARN] [1667373571.995027436]: "camera_link" passed to lookupTransform argument target_frame does not exist. 
...
[ WARN] [1667373654.288250251]: "camera_link" passed to lookupTransform argument target_frame does not exist. 

Saving keyframe trajectory to KeyFrameTrajectory.txt ...

trajectory saved!
shutting down processing monitor...
... shutting down processing monitor complete
done
```
*************************************************************
**Rostopic list:**

```
/clicked_point
/initialpose
/move_base_simple/goal
/orb_slam2_mono/debug_image
/orb_slam2_mono/debug_image/compressed
/orb_slam2_mono/debug_image/compressed/parameter_descriptions
/orb_slam2_mono/debug_image/compressed/parameter_updates
/orb_slam2_mono/debug_image/compressedDepth
/orb_slam2_mono/debug_image/compressedDepth/parameter_descriptions
/orb_slam2_mono/debug_image/compressedDepth/parameter_updates
/orb_slam2_mono/debug_image/theora
/orb_slam2_mono/debug_image/theora/parameter_descriptions
/orb_slam2_mono/debug_image/theora/parameter_updates
/orb_slam2_mono/gba_running
/orb_slam2_mono/map_points
/orb_slam2_mono/parameter_descriptions
/orb_slam2_mono/parameter_updates
/orb_slam2_mono/pose
/rosout
/rosout_agg
/tf
/tf_static
/webcam/camera_info
/webcam/image_raw
/webcam/image_raw/compressed
/webcam/image_raw/compressed/parameter_descriptions
/webcam/image_raw/compressed/parameter_updates
/webcam/image_raw/compressedDepth
/webcam/image_raw/compressedDepth/parameter_descriptions
/webcam/image_raw/compressedDepth/parameter_updates
/webcam/image_raw/theora
/webcam/image_raw/theora/parameter_descriptions
/webcam/image_raw/theora/parameter_updates
/webcam/webcam_image_view/output
/webcam/webcam_image_view/parameter_descriptions
/webcam/webcam_image_view/parameter_updates
/webcam/webcam_stream/parameter_descriptions
/webcam/webcam_stream/parameter_updates
```

qiuliangcheng, 2023-04-17:
Have you found a solution yet? I also encountered the same problem。thank you!

MagdaZal, 2023-04-18:
Hi @qiuliangcheng. To be honest I don't remember the solution. I have made so many changes in program since November... But I recommend you to check your tf graph (https://www.ros.org/reps/rep-0105.html) and add some static/dynamic transforms between frames or use repository which do the transformation. Additionally adding an INS sensor should help